lm-sensors/lm-sensors #469: sensor-detect is not detecting the fan rpm speed acpi device

rabipaikaray, 2023-11-10:
Hi,
I am using a intel alder lake (ADL-S) Platform

It has a fan device and I could read the fan speed from the sys file
$ cat /sys/bus/acpi/devices/INTC1048:00/fan_speed_rpm
1146

When I am running sensor from lm_sensor it does not detect the fan device
Out of the command is pasted below.
$ sudo sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +31.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 4:        +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 8:        +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 12:       +29.0°C  (high = +80.0°C, crit = +100.0°C)
Core 16:       +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 20:       +29.0°C  (high = +80.0°C, crit = +100.0°C)
Core 24:       +26.0°C  (high = +80.0°C, crit = +100.0°C)
Core 28:       +29.0°C  (high = +80.0°C, crit = +100.0°C)
Core 36:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 37:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 38:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 39:       +28.0°C  (high = +80.0°C, crit = +100.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +32.0°C  (crit = +105.0°C)

nvme-pci-0200
Adapter: PCI adapter
Composite:    +27.9°C  (low  =  -5.2°C, high = +79.8°C)
                       (crit = +84.8°C)
$ **sudo inxi -f**
CPU:
  Info: 12-core (8-mt/4-st) model: 12th Gen Intel Core i7-12700E bits: 64
    type: MST AMCP cache: L2: 12 MiB
  Speed (MHz): avg: 815 min/max: 800/4800:3600 cores: 1: 833 2: 891 3: 800
    4: 834 5: 800 6: 854 7: 799 8: 856 9: 800 10: 800 11: 801 12: 841 13: 800
    14: 800 15: 800 16: 800 17: 800 18: 800 19: 800 20: 800
  Flags: 3dnowprefetch abm acpi adx aes aperfmperf apic arat
    arch_capabilities arch_lbr arch_perfmon art avx avx2 avx_vnni bmi1 bmi2 bts
    clflush clflushopt clwb cmov constant_tsc cpuid cpuid_fault cx16 cx8 de
    ds_cpl dtes64 dtherm dts epb ept ept_ad erms est f16c flexpriority
    flush_l1d fma fpu fsgsbase fsrm fxsr gfni ht hwp hwp_act_window hwp_epp
    hwp_notify hwp_pkg_req ibpb ibrs ibrs_enhanced ida intel_pt invpcid
    lahf_lm lm mca mce md_clear mmx monitor movbe movdir64b movdiri msr mtrr
    nonstop_tsc nopl nx ospke pae pat pbe pclmulqdq pconfig pdcm pdpe1gb pebs
    pge pku pln pni popcnt pse pse36 pts rdpid rdrand rdseed rdtscp rep_good
    sdbg sep serialize sha_ni smap smep smx split_lock_detect ss ssbd sse
    sse2 sse4_1 sse4_2 ssse3 stibp syscall tm tm2 tme tpr_shadow tsc
    tsc_adjust tsc_deadline_timer tsc_known_freq umip vaes vme vmx vnmi
    vpclmulqdq vpid waitpkg x2apic xgetbv1 xsave xsavec xsaveopt xsaves
    xtopology xtpr

Looking for a help here why lm sensor is not detecting the fan speed rpm device.
Is there any specific steps I am missing here?
If I want to include this acpi INTC1048 to lm sensor how can I add it.

rabipaikaray, 2023-11-10:
sensors-detect output
```c
$ sudo sensors-detect
# sensors-detect version 3.6.0
# System: Intel Corporation Alder Lake Client Platform [0.1]
# Board: Intel Corporation AlderLake-S ADP-S DDR4 UDIMM CRB
# Kernel: 5.15.0-1043-intel-iotg x86_64
# Processor: 12th Gen Intel(R) Core(TM) i7-12700E (6/151/2)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): yes
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): yes
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     Yes
Found unknown chip with ID 0xfe00
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): yes
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): yes
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
[... 122 lines cut ...]
Core 12:       +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 16:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
Core 20:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 24:       +26.0°C  (high = +80.0°C, crit = +100.0°C)
Core 28:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 36:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
Core 37:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
Core 38:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
Core 39:       +30.0°C  (high = +80.0°C, crit = +100.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +32.0°C  (crit = +105.0°C)

nvme-pci-0200
Adapter: PCI adapter
Composite:    +27.9°C  (low  =  -5.2°C, high = +79.8°C)
                       (crit = +84.8°C)

``` 
ACPI fan device fan_speed_rpm output
```c
cat /sys/bus/acpi/devices/INTC1048:00/fan_speed_rpm
1145
```

Wer-Wolf, 2023-11-14:
The reason for this is that the ACPI fan driver does not implement the standard hwmon interface.

If you are willing to test some modified kernel modules, i can fix that.

rabipaikaray, 2023-11-14:
Hi @Wer-Wolf 
Could you let me know more about this?
_If you are willing to test some modified kernel modules, i can fix that._

Which Kernel module we need to change in order to lm_sensor to  list the device. As I understand the the fan.c which is part of acpi kernel code.

https://github.com/GalliumOS/linux/blob/master/drivers/acpi/fan.c
Let me know what changes we can bring here to test this?

Wer-Wolf, 2023-11-14:
I assume that the driver is build as a module, so you dont have to rebuild your whole kernel.

Wer-Wolf, 2023-11-14:
Could you send me a dump of your ACPI tables using `acpidump`?

rabipaikaray, 2023-11-14:
Hi @Wer-Wolf 
It is not built as module. You can suggest me the change and let me do the modification and test.

rabipaikaray, 2023-11-14:
Hi @Wer-Wolf 
[acpidump.txt](https://github.com/lm-sensors/lm-sensors/files/13347785/acpidump.txt)
Attached the acpidump here.

Wer-Wolf, 2023-11-14:
The fan driver needs to be modified to register a hwmon device, see [here](https://www.kernel.org/doc/html/latest/hwmon/) for details.

Wer-Wolf, 2023-11-16:
[0001-ACPI-fan-Add-hwmon-support.patch.txt](https://github.com/lm-sensors/lm-sensors/files/13383903/0001-ACPI-fan-Add-hwmon-support.patch.txt)
Can you apply this kernel patch and check if `sensors` now displays an `acpi_fan` device?

Wer-Wolf, 2023-12-18:
Any progress?

rabipaikaray, 2023-12-19:
Hi @Wer-Wolf 
Thanks for the patch.
Not able to do the changes as the priority of doing the kernel changes changed.
I am reading now directly from the sysfs entries to get the fan speed rather using the lm-sensor.

Wer-Wolf, 2024-04-04:
Ok