lm-sensors/lm-sensors #470: Armbian N/A cpu temterature

a9fm, 2023-11-18:
$ sensors
rpi_volt-isa-0000
Adapter: ISA adapter
in0:              N/A

 $ uname -a
Linux rpi4b 6.1.50-current-bcm2711 #7 SMP PREEMPT Fri Sep  1 12:43:26 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux

                                 root@rpi4b
                                 ----------
      █ █ █ █ █ █ █ █ █ █ █      OS: Armbian (23.8.1) aarch64
     ███████████████████████     Host: Raspberry Pi 4 Model B Rev 1.4
   ▄▄██                   ██▄▄   Kernel: 6.1.50-current-bcm2711
   ▄▄██    ███████████    ██▄▄   Uptime: 3 mins
   ▄▄██   ██         ██   ██▄▄   Packages: 1815 (dpkg)
   ▄▄██   ██         ██   ██▄▄   Shell: zsh 5.8.1
   ▄▄██   ██         ██   ██▄▄   Resolution: 1920x1080
   ▄▄██   █████████████   ██▄▄   CPU: BCM2835 (4) @ 1.800GHz
   ▄▄██   ██         ██   ██▄▄   Memory: 1507MiB / 7633MiB
   ▄▄██   ██         ██   ██▄▄
   ▄▄██   ██         ██   ██▄▄
   ▄▄██                   ██▄▄
     ███████████████████████
      █ █ █ █ █ █ █ █ █ █ █

Wer-Wolf, 2023-11-21:
Could it be the thermal subsystem is builtin and the hwmon subsystem is build as an module? In this case, the thermal driver for the cpu cannot setup an hwmon device, so it does not show up when using sensors.

a9fm, 2023-11-24:
How to check that?

Wer-Wolf, 2023-11-24:
Execute `modinfo thermal` and `modinfo hwmon`. If the filename of one of them is `(builtin)`, then it is builtin, else it is built as an module.