lm-sensors/lm-sensors #471: Add support

mohamad-hasan-sohan-ajini, 2023-11-27:
machine info.:

> (venv) aj@aj-G5:~$ inxi -M
Machine:
  Type: Desktop System: HP product: HP EliteDesk 800 G5 SFF v: N/A
    serial: <superuser required>
  Mobo: HP model: 8592 v: KBC Version 08.97.00 serial: <superuser required>
    UEFI: HP v: R01 Ver. 02.05.01 date: 04/27/2020

sensors-detect info.:

> Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family \`National Semiconductor/ITE\'...               Yes
Found unknown chip with ID 0x1f11
Probing for Super-I/O at 0x4e/0x4f
Trying family \`National Semiconductor/ITE\'...               No
Trying family \`SMSC\'...                                     No
Trying family \`VIA/Winbond/Nuvoton/Fintek\'...               No
Trying family \`ITE\'...                                      No

isadump info.:

> aj@aj-G5:~$ sudo isadump -k 0x87,0x01,0x55,0x55 0x2e 0x2f 7
WARNING! Running this program can cause system crashes, data loss and worse!
I will probe address register 0x2e and data register 0x2f.
Probing bank 7 using bank register 0x07.
Continue? [Y/n]  
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00: 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 00 
10: 02 24 00 00 80 1d 53 2b 1f 00 b0 1e a8 08 00 00 
20: 1f 11 7b b6 12 24 00 02 90 00 20 00 00 00 00 00 
30: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
50: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
60: 0a 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
70: 00 03 00 00 04 04 00 00 00 00 00 00 00 00 00 00 
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
f0: 50 04 00 00 00 00 00 00 01 00 00 00 00 00 00 00 

modprobe and dmesg info.:

> aj@aj-G5:\~$ sudo modprobe it87
modprobe: ERROR: could not insert 'it87': No such device
aj@aj-G5:\~$ sudo dmesg | grep it87
[ 1186.535216] it87: it87 driver version v1.0-48-g40bec4b

sensors info.:

> (venv) aj@aj-G5:~$ sensors
pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +41.0°C  
nvme-pci-0200
Adapter: PCI adapter
Composite:    +29.9°C  (low  =  -0.1°C, high = +117.8°C)
                       (crit = +149.8°C)
acpitz-acpi-0
Adapter: ACPI interface
temp1:        +30.0°C  
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +36.0°C  (high = +82.0°C, crit = +100.0°C)
Core 0:        +34.0°C  (high = +82.0°C, crit = +100.0°C)
Core 1:        +36.0°C  (high = +82.0°C, crit = +100.0°C)
Core 2:        +34.0°C  (high = +82.0°C, crit = +100.0°C)
Core 3:        +35.0°C  (high = +82.0°C, crit = +100.0°C)
Core 4:        +33.0°C  (high = +82.0°C, crit = +100.0°C)
Core 5:        +34.0°C  (high = +82.0°C, crit = +100.0°C)
ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

Also was not able to force loading driver: `sudo modprobe -r it87 && modprobe it87 force_id=0x1f11`. 
Please help me with this error.

Thanks in advance

Wer-Wolf, 2023-12-09:
We sadly cannot help you, the datasheets necessary for adding support for your superio chip are likely not publicly available.

Wer-Wolf, 2023-12-11:
However, we can check if an alternative interface to read sensor values exists. Can you share the output of `acpidump`?

alium, 2024-03-18:
Hi, i have  AMD Ryzen 5 PRO 4650GE with Radeon Graphics, HP ProDesk 405 G6 Desktop Mini PC, version  SBKPF, DMI: HP HP ProDesk 405 G6 Desktop Mini PC/872E, BIOS S27 Ver. 02.14.00 09/21/2023

with same sensor chip: Trying family `National Semiconductor/ITE'... Yes Found unknown chip with ID 0x1f11. 
[acpidump.txt](https://github.com/lm-sensors/lm-sensors/files/14641705/acpidump.txt)

You asked for acpidump, so here is it

Wer-Wolf, 2024-03-18:
It seems this HP machine could be supported by the `hp-wmi-sensors` driver.
Can you check that the driver is present an loaded?

alium, 2024-03-18:
no, is not. i must load hp_wmi_sensors manually 

`[root@renoir alois]# lsmod | grep -i hp_wmi
hp_wmi                 32768  0
platform_profile       12288  1 hp_wmi
sparse_keymap          12288  1 hp_wmi
rfkill                 40960  8 hp_wmi,bluetooth,cfg80211
wmi                    36864  3 hp_wmi,video,wmi_bmof`

`[root@renoir alois]# modprobe hp_wmi_sensors`

`[root@renoir alois]# lsmod | grep -i hp_wmi_sensors
hp_wmi_sensors         36864  0
wmi                    36864  4 hp_wmi,video,hp_wmi_sensors,wmi_bmof`

alium, 2024-03-18:
but this is new entry , if i run `sensors`:

[root@renoir alois]# sensors
>hp_wmi_sensors-virtual-0
>Adapter: Virtual device
>CPU Fan Speed:       1524 RPM
>:                       0 RPM
>CPU Temperature:      +41.0°C  
>Chassis Temperature:  +25.0°C

Wer-Wolf, 2024-03-18:
Oh, it seems that the driver does not support autoloading. I will submit a patch upstream which should fix that.

kangtastic, 2024-03-21:
Hi @alium, hp-wmi-sensors driver author here. Sorry about the lack of autoloading so far. I intentionally left it out originally, and I knew someone would eventually notice :grin:

[Here](https://lore.kernel.org/linux-hwmon/3e069b7a-fa22-453b-a507-dccc48eb60a3@gmx.de/T/) is the discussion that resulted from @Wer-Wolf submitting the patch he mentioned.

By the way, I noticed your `sensors` output has a missing sensor name:

> : 0 RPM

I decompiled your `acpidump.txt` to ACPI Source Language (ASL) - [acpidump.asl.txt](https://github.com/lm-sensors/lm-sensors/files/14681464/acpidump.asl.txt) - and I think the cause is at line 10550:

> If (((Zero && (FNC4 != 0xFF)) && (FNC4 < 0x0D)))

which I'm guessing should have been:

> If (((Ones && (FNC4 != 0xFF)) && (FNC4 < 0x0D)))

Unfortunately it looks like this is probably a BIOS bug. There's nothing we can do about it, nor an obvious way to tell which sensor it's supposed to be.

If you're so inclined, the official performance monitoring tool you could use to check that it also shows up without a sensor name in Windows is [HP Performance Advisor](https://www.hp.com/us-en/workstations/performance-advisor.html). No need to do this, though, we already know the results.

alium, 2024-03-21:
hi @kangtastic, thank you very much for the message. I have read the discussion. Yes, HP probably have a bug in the BIOS (sometimes dmesg tells me this and also says I have 2 duplicate WMI GUIDs.
```
[    4.227646] wmi_bus wmi_bus-PNP0C14:00: [Firmware Info]: DF4E63B6-3BBC-4858-9737-C74F82F821F3 has zero instances
[    4.227681] wmi_bus wmi_bus-PNP0C14:00: [Firmware Info]: 8F1F6436-9F42-42C8-BADC-0E9424F20C9A has zero instances
[    4.230001] acpi PNP0C14:02: duplicate WMI GUID 2B814318-4BE8-4707-9D84-A190A859B5D0 (first instance was on PNP0C14:00)
[    4.230018] acpi PNP0C14:02: duplicate WMI GUID 41227C2D-80E1-423F-8B8E-87E32755A0EB (first instance was on PNP0C14:00)
[    4.230022] wmi_bus wmi_bus-PNP0C14:02: WQZZ data block query control method not found
```

 I understand that it's probably hard to do anything about it.
 
However, if @Wer-Wolf  will supply some patches, I can at least test them on my machine, if there is interest. 

And thanks for your work and time!

Wer-Wolf, 2024-03-21:
The ACPI tables contains a couple of invalid WMI devices, those will automatically be ignored by the WMI subsystem.

The errors concerning the duplicated GUIDs are harmless and will be resolved in kernel version 6.9 (the next release candidate).