lm-sensors/lm-sensors #473: lm_sensors.service: Referenced but unset environment variable evaluates to an empty string: BUS_MODULES

ilf, 2023-12-10:
systemd 254 includes this change:

> When resolving environment variables we currently silently resolve unset and invalid environment variables to empty strings.

https://github.com/systemd/systemd/commit/f331434d13488425ccd8485327085d15f8f92aea

This causes https://github.com/lm-sensors/lm-sensors/blob/master/prog/init/lm_sensors.service to say:

> lm_sensors.service: Referenced but unset environment variable evaluates to an empty string: BUS_MODULES

See https://bugs.archlinux.org/task/79273

mpan-pl, 2024-02-12:
`BUS_MODULES` should be defined in `/etc/conf.d/lm_sensors` (`EnvironmentFile` loads it). The file is user-created, not a part of the upstream deployment.

The user [_may_ generate the file](https://github.com/lm-sensors/lm-sensors/blob/42f240d2a457834bcbdf4dc8b57237f97b5f5854/prog/detect/sensors-detect#L7455-L7456) with `sensors-detect`. So perhaps this script might be changed to unconditionally define `BUS_MODULES`, even if empty.

ManuLinares, 2024-05-03:
To fix this, and indeed could be a security issue, maybe it is a good idea to evaluate the variables.

https://github.com/lm-sensors/lm-sensors/blob/master/prog/init/lm_sensors.service

```
...
ExecStart=-/sbin/modprobe -qab ${BUS_MODULES- } ${HWMON_MODULES- }
ExecStart=/usr/bin/sensors -s
ExecStop=-/sbin/modprobe -qabr ${BUS_MODULES- } ${HWMON_MODULES- }
...
```

This stops the warning. Just an idea.

genodeftest, 2024-07-20:
> `BUS_MODULES` should be defined in `/etc/conf.d/lm_sensors` (`EnvironmentFile` loads it). The file is user-created, not a part of the upstream deployment.

On my machine¹, there is no `/etc/conf.d/lm_sensors` file. The config files/folders are:
* `/etc/sensors.d` directory, which is empty.
* `/etc/sensors3.conf`, containing a lot of chip and label definitions
* `/etc/sysconfig/lm_sensors`, the environment file passed to lm_sensors as systemd `EnvironmentFile`. This file was created by `sensors-detect` 3.6.0. This file contains a definition of `HWMON_MODULES`, but it does not contain a definition of `BUS_MODULES` (regenerating the file only changes the date&time):
```
# Generated by sensors-detect on Thu Mar  2 14:57:53 2023
# This file is sourced by /etc/init.d/lm_sensors and defines the modules to
# be loaded/unloaded.
#
# The format of this file is a shell script that simply defines variables:
# HWMON_MODULES for hardware monitoring driver modules, and optionally
# BUS_MODULES for any required bus driver module (for example for I2C or SPI).

HWMON_MODULES="coretemp w83627ehf"
```

> The user [_may_ generate the file](https://github.com/lm-sensors/lm-sensors/blob/42f240d2a457834bcbdf4dc8b57237f97b5f5854/prog/detect/sensors-detect#L7455-L7456) with `sensors-detect`. So perhaps this script might be changed to unconditionally define `BUS_MODULES`, even if empty.

+1, sounds like this is the change needed.

_____
¹ Fedora 40, lm_sensors 3.6.0 (-18.fc40)

ManuLinares, 2024-11-13:
Checking if, a year later, anyone has found the best way to fix this