lm-sensors/lm-sensors #474: ASUS ROG STRIX X670E-A Gaming Wifi

StoffePro, 2023-12-14:
Hello. Running sensors-detect on my ASUS ROG STRIX X670E-A Gaming Wifi does absolutely nothing:

```
Sorry, no sensors were detected.
```

I'm running Arch, and my kernel version is 6.6.7. Everything is fully up to date.

I realize I should probably provide a ton more info, but this is way out of my comfort zone, so I don't know what's relevant.

Any pointers would be greatly appreciated.

Wer-Wolf, 2023-12-28:
Is secure boot enabled?

StoffePro, 2024-01-05:
> Is secure boot enabled?

No.

Wer-Wolf, 2024-01-05:
Maybe we need to whitelist `asus-ec-sensors` for your motherboard. Would you be able to compile a kernel module?

Wer-Wolf, 2024-01-05:
Also could you share the output of `acpidump`?

StoffePro, 2024-01-06:
> Maybe we need to whitelist `asus-ec-sensors` for your motherboard. Would you be able to compile a kernel module?

Sure.

StoffePro, 2024-01-06:
> Also could you share the output of `acpidump`?

Gladly: https://gist.github.com/StoffePro/645674d3430bb184bf12aacce28314e7

Wer-Wolf, 2024-01-06:
Can you also tell me what sensors are displayed inside you motherboards BIOS menu?

StoffePro, 2024-01-12:
> Can you also tell me what sensors are displayed inside you motherboards BIOS menu?

Sorry about the delay, I thought I had answered, must have forgotten to press the button.

What exactly am I looking for here? I couldn't find any references to actual chips or anything, do you just want a list of all the temps and fan speeds etc?

Wer-Wolf, 2024-01-12:
Just a list of all the temps and fan speeds.

StoffePro, 2024-01-15:
)

> Just a list of all the temps and fan speeds.

- Monitor
  - Temperature Monitor
    - CPU Temperature
    - CPU Package Temperature
    - MotherBoard Temperature
    - VRM Temperature
    - Chipset 1 Temperature
    - Chipset 2 Temperature
    - T_Sensor Temperature
    - DIMM A1 Temperature
    - DIMM A2 Temperature
    - DIMM B1 Temperature
    - DIMM B2 Temperature
  - Fan Speed Monitor
    - CPU Fan Speed
    - CPU Optional Fan Speed
    - Chassis Fan 1 Speed
    - Chassis Fan 2 Speed
    - Chassis Fan 3 Speed
    - Chassis Fan 4 Speed
    - Chassis Fan 5 Speed
    - AIO Pump Speed
  - Voltage Monitor
    - CPU Core Voltage
    - 12V Voltage
    - 5V Voltage
    - 3.3V Voltage
    - CPU VDDIO / MC Voltage

There's also a bunch of fan profile stuff under there. Is that interesting?

Wer-Wolf, 2024-01-20:
The fan profile stuff is supported by a different driver.

I asked the maintainer of the `asus-ec-sensors` kernel driver for advise:

>Hi Armin,
>
>The first part of
>https://github.com/zeule/asus-ec-sensors?tab=readme-ov-file#adding-a-new-motherboard
>contains some useful information (up to the BREC function, which is
>outdated). Basically, you can find a similar board already included in
>the driver, copy and change its definition. I guess for X670E we have
>currently only board_info_crosshair_x670e_hero, so copy that. You need
>to find out T_Sensor address and add it to the sensors_family_amd_600
>array. To find sensor addresses, you can dump EC registers using the
>ec_sys kernel module. Please, do not hesitate to ask further
>questions, although I will again be somewhat slow to respond.
>
>Cheers,
>Eugene

Do you thing you can do the driver modification yourself?

SaigyoujiYuyuko233, 2025-07-28:
I got the same motherboard and no sensor detected output. I'm running fedora42 w/ ``6.15.4-200.fc42.x86_64+rt #1 SMP PREEMPT_RT Sat Jul  5 12:21:28 EDT 2025 x86_64 GNU/Linux`` (self-compiled with realtime support) and everything is fully up to date, too.

Just follow up with what I got, thanks!