lm-sensors/lm-sensors #475: Missing fan information from MSI GF63-12VF

Kalabaza, 2023-12-15:
The data from the fan speed is not available in lm-sensors.

```shell
user@Thin-GF63-12VF:~$ sudo sensors-detect 
# sensors-detect version 3.6.0
# System: Micro-Star International Co., Ltd. Thin GF63 12VF [REV:1.0] (laptop)
# Board: Micro-Star International Co., Ltd. MS-16R8
# Kernel: 6.2.0-39-generic x86_64
# Processor: 12th Gen Intel(R) Core(TM) i7-12650H (6/154/3)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
/dev/port: Operation not permitted

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
/dev/port: Operation not permitted

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 8086:51a3 at 0000:00:1f.4.
Sorry, no supported PCI bus adapters found.

Next adapter: SMBus I801 adapter at efa0 (i2c-0)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x18
Handled by driver `jc42' (already loaded), chip type `jc42'
Client found at address 0x1a
Handled by driver `jc42' (already loaded), chip type `jc42'
Client found at address 0x50
Handled by driver `ee1004' (already loaded), chip type `ee1004'
    (note: this is probably NOT a sensor chip!)
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)

Next adapter: Synopsys DesignWare I2C adapter (i2c-1)
Do you want to scan it? (YES/no/selectively): 
Adapter doesn't support all probing functions.
Some addresses won't be probed.

Next adapter: i915 gmbus dpa (i2c-2)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpb (i2c-3)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpc (i2c-4)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc1 (i2c-5)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc2 (i2c-6)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc3 (i2c-7)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc4 (i2c-8)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc5 (i2c-9)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc6 (i2c-10)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 1 at 1:00.0 (i2c-11)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 5 at 1:00.0 (i2c-12)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AUX A/DDI A/PHY A (i2c-13)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AUX USBC1/DDI TC1/PHY TC1 (i2c-14)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AUX USBC2/DDI TC2/PHY TC2 (i2c-15)
Do you want to scan it? (yes/NO/selectively): 

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Driver `jc42':
  * Bus `SMBus I801 adapter at efa0'
    Busdriver `i2c_i801', I2C address 0x18
    Chip `jc42' (confidence: 6)
  * Bus `SMBus I801 adapter at efa0'
    Busdriver `i2c_i801', I2C address 0x1a
    Chip `jc42' (confidence: 6)

To load everything that is needed, add this to /etc/modules:
#----cut here----
# Chip drivers
coretemp
jc42
#----cut here----
If you have some drivers built into your kernel, the list above will
contain too many modules. Skip the appropriate ones!

Do you want to add these lines automatically to /etc/modules? (yes/NO)

Unloading cpuid... OK
```

Output from sensors:

```shell
user@Thin-GF63-12VF:~$ sensors
jc42-i2c-0-1a
Adapter: SMBus I801 adapter at efa0
temp1:        +44.2°C  (low  =  +0.0°C)                  ALARM (HIGH, CRIT)
                       (high =  +0.0°C, hyst =  +0.0°C)
                       (crit =  +0.0°C, hyst =  +0.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +47.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +43.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +47.0°C  (high = +100.0°C, crit = +100.0°C)
Core 8:        +41.0°C  (high = +100.0°C, crit = +100.0°C)
Core 12:       +44.0°C  (high = +100.0°C, crit = +100.0°C)
Core 16:       +42.0°C  (high = +100.0°C, crit = +100.0°C)
Core 20:       +42.0°C  (high = +100.0°C, crit = +100.0°C)
Core 28:       +41.0°C  (high = +100.0°C, crit = +100.0°C)
Core 29:       +41.0°C  (high = +100.0°C, crit = +100.0°C)
Core 30:       +41.0°C  (high = +100.0°C, crit = +100.0°C)
Core 31:       +41.0°C  (high = +100.0°C, crit = +100.0°C)

BAT1-acpi-0
Adapter: ACPI interface
in0:          12.74 V  
curr1:         0.00 A  

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +59.0°C  

jc42-i2c-0-18
Adapter: SMBus I801 adapter at efa0
temp1:        +44.8°C  (low  =  +0.0°C)                  ALARM (HIGH, CRIT)
                       (high =  +0.0°C, hyst =  +0.0°C)
                       (crit =  +0.0°C, hyst =  +0.0°C)

nvme-pci-0300
Adapter: PCI adapter
Composite:    +39.9°C  (low  = -20.1°C, high = +89.8°C)
                       (crit = +94.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +51.0°C  (crit = +100.0°C)
```

According to the error in sensors-detect, this was not detected correctly:

```shell
user@Thin-GF63-12VF:~$ lspci -nn | grep 1f.4
00:1f.4 SMBus [0c05]: Intel Corporation Alder Lake PCH-P SMBus Host Controller [8086:51a3] (rev 01)
```

Any help will be appreciated.

Wer-Wolf, 2023-12-28:
Can you share the output of `acpidump`?

Kalabaza, 2024-01-12:
Sure, here is the output file.
[acpidump.txt](https://github.com/lm-sensors/lm-sensors/files/13917232/acpidump.txt)

Wer-Wolf, 2024-01-13:
Can you also share the content of `/sys/bus/wmi/device/05901221-D566-11D1-B2F0-00A0C9062910*/bmof`?

Kalabaza, 2024-01-13:
Here is the other data that you wanted, and sure thing I can test that driver, after reporting this problem I switched to a locally compiled kernel, so now I'm using this:

Kernel: 6.2.16 x86_64

Just let me know if you need anymore information from the computer.

[bmof.zip](https://github.com/lm-sensors/lm-sensors/files/13928368/bmof.zip)

Wer-Wolf, 2024-01-13:
Ok, i will try to get a prototype working over the weekend.

Wer-Wolf, 2024-01-14:
It turned out a driver for this device was going to be released "soon" back in 2014 by Intel itself, but this never happend.

I will try to get this driver upstreamed before i start writing my own.

Wer-Wolf, 2024-01-16:
While we are waiting for a response from Intel, could you share the content of `/sys/bus/hid/devices/*/report_descriptor`?

Kalabaza, 2024-01-16:
Sure, here are is the content of there:

$ ls -l /sys/bus/hid/devices/00*
lrwxrwxrwx 1 root root 0 Jan 16 22:22 /sys/bus/hid/devices/0003:046D:C542.0002 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:046D:C542.0002
lrwxrwxrwx 1 root root 0 Jan 16 22:22 /sys/bus/hid/devices/0018:04F3:30AA.0001 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-1/i2c-MSNB0001:00/0018:04F3:30AA.0001

[report_descriptor.zip](https://github.com/lm-sensors/lm-sensors/files/13956070/report_descriptor.zip)

Wer-Wolf, 2024-01-18:
I found another possible interface for reading the fan speed over the WMI interface.

Can you dualboot into Windows? If yes, then install the MSI Dragon Center software and check if you can read the fan speed this way.
If that is the case, send me all files which are listed by `grep -r FOMB`.

Kalabaza, 2024-01-18:
I can dualboot into Windows 11, but the MSI Dragon Center is no longer available with this processor model, but I do have the MSI Center, which is supposed to be the next version of that app.

So, I can see one fan speed, which makes sense since the computer only has one fan.

[image]

The only file that I could find that contains that "FOMB" string is located in the folder "MSI NBFoundation Service" and is the dll "msiapcfg.dll", I hope that this is what you need, otherwise let me know.

[msiapcfg.zip](https://github.com/lm-sensors/lm-sensors/files/13982659/msiapcfg.zip)

Wer-Wolf, 2024-01-19:
This was exactly that i was searching for, the WMI interface in question does have a `Get_Fan` and `Set_Fan` method.
I will develop a small prototype driver to gather some system data for further reverse engineering.

Wer-Wolf, 2024-01-19:
Regarding the `INT340E` device:

Intel said its deprecated and claims to have no documentation regarding its interface.
It also seems that the device in your machine is not working, so i will go with the WMI-based interface.

Wer-Wolf, 2024-01-26:
Sorry for the delay, it took me some time to come up with a good debug interface.

[msi-wmi-platform.zip](https://github.com/lm-sensors/lm-sensors/files/14062703/msi-wmi-platform.zip)

After building and loading this module, you can execute WMI methods by writing the binary buffer into `/sys/kernel/debug/msi-wmi-platform-ABBC0F6E-8EA1-11d1-00A0-C90629100000/<method_name>` and reading back a binary buffer containing the result.

Can you write a 32 byte buffer starting with either `0x00` or `0x01` into `get_fan` and send me the returned binary buffer?

Please note that the module might be buggy, so prepare for kernel panics!

Kalabaza, 2024-01-28:
Hello, so the other day I tried to use the module but it was not working correctly, I didn't have time that day to debug it, so I was able to do that today.

There are two errors in the driver, I solve them and now I can run it:

1- In the write function, the data is not actually located in fp->private_data but in fp->f_inode->i_private. This was causing that the sent data from the user side was not even used.
2- In the show function, seq_write expects to write less data than the maximum available in the seq_file, in this case if you want to write the 32 bytes, a custom function has to be used or write one byte less than the maximum.

```
int seq_write(struct seq_file *seq, const void *data, size_t len)
{
	// in a 32 bytes buffer, only 31 can be written
	if (seq->count + len < seq->size) {
		memcpy(seq->buf + seq->count, data, len);
		seq->count += len;
		return 0;
	}
	seq_set_overflow(seq);
	return -1;
}
```

[msi-wmi-platform.zip](https://github.com/lm-sensors/lm-sensors/files/14076038/msi-wmi-platform.zip)

With those two fixes applied, the query is executed correctly in the wmi platform and I get this (didn't know where you wanted that 0x01 endianness, so I executed both cases):

**writing to get_fan**
`echo -n -e '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' > get_fan`
**content of get_fan**
`01 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00`

**writing to get_fan**
`echo -n -e '\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' > get_fan`
**content of get_fan**
`01 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00`

**writing to get_fan**
`echo -n -e '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01' > get_fan`
**content of get_fan**
`01 00 C8 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00`

Wer-Wolf, 2024-01-28:
Thanks, for figuring out the errors i made.

How fast was you fan spinning when you where executing the last write?
Somehow you endianness of the buffer seems to be mixed up, i will investigate that.

Can you check what happens if you pass a buffer starting with `0x02`?

Wer-Wolf, 2024-01-28:
[msi-wmi-platform.zip](https://github.com/lm-sensors/lm-sensors/files/14076343/msi-wmi-platform.zip)

I tried to fix the errors you found:
- in `msi_wmi_platform_write`, the private data is stored inside the `seq_file`, not inside the `file` itself
- `seq_write` seems to use the last byte inside the buffer to detect buffer overflows, so passing a larger buffer should solve the problem

Can you check if the fixes are correct?

Wer-Wolf, 2024-01-28:
Can you also send me the whole `MSI NBFoundation Service` directory where you found the `msiapcfg.dll`?

Kalabaza, 2024-01-28:
Ok, with 0x02 I'm getting this:

`01 00 00 28  32 3C 46 50  55 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00`

And, the fix of the two issues was already there, in my post, I modified the file and just included the fixed c file in the zip.

But about your changes, one is right and one wrong.

1. That change in the write function is working, seems that the private data pointer is located in two places: 

My solution:
`struct msi_wmi_platform_debugfs_data *data = fp->f_inode->i_private;`

Your solution:
```
struct seq_file *seq = fp->private_data;
struct msi_wmi_platform_debugfs_data *data = seq->private;
```

So, whichever you use is fine.

2. Changing the buffer like that will actually cause an overflow, the seq->count is modified after doing a memcpy and it is incremented by the length, so if you call a second time seq_write, the start position of the memcpy uses the buffer address plus that seq->count, so the start will be one byte out of place.

So, I just copy the code of seq_write from the local kernel files and modify the comparisson to make it work:

`if (seq->count + len <= seq->size)`

Kalabaza, 2024-01-28:
Here is the folder in where I found that dll in Windows, let me know if you need something more.

[MSI NBFoundation Service.zip](https://github.com/lm-sensors/lm-sensors/files/14076612/MSI.NBFoundation.Service.zip)

Wer-Wolf, 2024-01-28:
With "passing a larger buffer" i meant passing a larger buffer size to `single_open_size()`.

Wer-Wolf, 2024-01-28:
The files you send me contain much information about the WMI interface. I will need some time to disassemble them.

Wer-Wolf, 2024-01-28:
After some studying, i have some more task for you. Can you send me the return of `get_ec` and `get_wmi` and the full MSI Center application directory?

Kalabaza, 2024-01-28:
Sure, here is the output of get_ec:

`01 81 31 36  52 38 49 4D  53 31 2E 31  30 37 30 35  32 34 32 30  32 33 31 35  3A 31 32 3A  35 30 00 00`

And this is from get_wmi:

`01 02 08 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00`

Kalabaza, 2024-01-28:
And here is the whole MSI Center folder (sorry but I had to use my google drive account, it is 188MB long, but I'm only able to upload up to 25MB here):

[MSI Center](https://drive.google.com/file/d/1a6VPWHquF77XMPBNt70jKR3dLtdZySsy/view?usp=sharing)

Wer-Wolf, 2024-01-28:
> Sure, here is the output of get_ec:
> 
> `01 81 31 36 52 38 49 4D 53 31 2E 31 30 37 30 35 32 34 32 30 32 33 31 35 3A 31 32 3A 35 30 00 00`
> 
> And this is from get_wmi:
> 
> `01 02 08 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

This means for the EC:
- Tiger Lake firmware interface is supported
- EC RAM version 0.1
- EC Firmware version `16R8IMS1.1070524202315:12:50`

For the WMI interface:
- WMI interface version `2.8`

Wer-Wolf, 2024-01-28:
I will take a look at the MSI Center files tomorrow. I hope that i can find out how to properly parse the fan sensor values.

Wer-Wolf, 2024-01-28:
Ok bad news it seems the raw byte stream is being send over a pipe to another process which i cannot disassemble.

Can you check how the values inside the buffers returned from `get_fan` change when the fan speed changes?

Kalabaza, 2024-01-28:
The values changes like this, when passing from no sound to heavy load and back to almost silent:

`01 00 00 28  32 3C 46 50  55 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00`

`01 28 00 28  32 3C 46 50  55 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00`

`01 55 00 28  32 3C 46 50  55 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00`

`01 3C 00 28  32 3C 46 50  55 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00`

`01 32 00 28  32 3C 46 50  55 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00`

`01 00 00 28  32 3C 46 50  55 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00`

Wer-Wolf, 2024-01-28:
What was the fan speed (as displayed in Windows) during this test?

Kalabaza, 2024-01-30:
If I do more or less the same test in Windows, the maximum fan speed is **3609 rpm** and from there to the minimum of **2341 rpm**. Then it simply shuts down, so no speed below that minimum.

Wer-Wolf, 2024-01-31:
I cannot see how those speed values come from buffer `0x02`. Can you check how the other buffers react to changes in fan speed?

Kalabaza, 2024-02-03:
Ok, here are some values with fan at the full speed for get_fan:

0x00
01 00 85 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00

0x01
01 55 00 28  32 3C 46 50  55 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00

0x02
01 46 00 28  32 3C 46 50  55 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00

0x03
00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00

The reset is just zeros.

Wer-Wolf, 2024-02-05:
Ok, i think 0x01 and 0x02 both show some kind of speed table with `0x55` being the current speed on 0x01 and possble speed values being `0x00`, `0x28`, `0x32` , `0x3C` ,`0x46` ,`0x50` and `0x55`.

I just do not know how to translate those values into RPM values. And i cannot think of a way to decode the buffer on 0x00.

Wer-Wolf, 2024-02-05:
I will try to contact someone who knows a bit about the MSI embedded controller interface. Can you check in the meantime if the fan is being controlled through performance states or by a smooth fan curve?

glpnk, 2024-02-05:
Oh, it is fun to know that I not one who tried to debug WMI2 interface. I also thought about creating VM which will emulate EC and WMI2 part of DSDT, because my laptop is using WMI1.

Is someone tried to contact MSI?

Did you find `CheckBackdoor` function inside app? In some cases, App just use raw access to EC, on WMI2 platform through `Get/Set_Data`, on WMI1 - WinIO driver

For test purpose of executing ACPI methods, I've used [`acpi_call` module](https://github.com/nix-community/acpi_call)

Wer-Wolf, 2024-02-05:
I found out that the WMI2 module executes `Get_EC` and `Get_WMI` to obtain the interface versions and check if the EC interface is supported.

Byte buffer format of `Get_EC`:

1. `0x01` for success, `0x00` for failure
2. Bitmap: Bit 8 is the "IsTigerLake" flag, bit 7 is the "EcRamChangedPage" flag, bit 6 and 5 are the EC RAM main version, remaining 4 bits are the EC RAM minor version
3. Firmware version string (UTF8, 28 bytes)

Wer-Wolf, 2024-02-05:
I was unable to further reverse engineer the app, since some modules are written in C++, which cannot be disassembled by dotPeek.

Wer-Wolf, 2024-02-05:
The WMI2 code was checking if the "IsTigerLake" flags is set and the major WMI interface version is greater than 2. Otherwise, errors would be logged.

glpnk, 2024-02-05:
[image]
Get_EC is basically a name + version code, if I remember it right

[image]
Get_WMI is just return version from DSDT device header
[image]

Wer-Wolf, 2024-02-05:
I can try to submit the `msi-wmi-platform` into the linux kernel staging area. It features a debugfs interface so this would benefit the reverse-engineering effort.

Wer-Wolf, 2024-02-05:
@Kalabaza Can you try and ask the MSI support about information on how to use the WMI interface? I doubt that they will share anything, but at least we can try.

Wer-Wolf, 2024-02-08:
I think i can now decode the fan speed buffers, i will publish an updated driver soon.

It will however take some time due to university exams.

Wer-Wolf, 2024-02-15:
[msi-wmi-platform.zip](https://github.com/lm-sensors/lm-sensors/files/14292452/msi-wmi-platform.zip)
This driver should now allow for reading the fan speed. Can you verify this?

Wer-Wolf, 2024-02-15:
Also it should print some debug information during startup. Can you share the output of `dmesg`?

Kalabaza, 2024-02-15:
Hello and first of all, I really appreciate the effort you are putting into this, thank you so much and you are making a good job.

So, I tried this latest driver and now I'm getting some readings in the **sensors** command, but definitely there are still some issues with the values:

// Speed when fan is off
```
$ sensors
msi_wmi_platform-virtual-0
Adapter: Virtual device
fan1:           FAULT
fan2:        12000 RPM
fan3:        8000 RPM
fan4:        6000 RPM
```

// Max speed of the fan (with max usage of GPU)
`fan1:        5647 RPM`

// Min speed of the fan (before turning off)
`fan1:        12000 RPM`

Now, going from the highest to the lowest speed is counter intuitive, so maybe the calculation of the speed is the problem:
```
fan1:        5647 RPM // Highest
fan1:        6857 RPM
fan1:        8000 RPM
fan1:        9600 RPM
fan1:        12000 RPM // Lowest
```

And this is the data that I got from **dmesg**:

```
[ 2509.267290] msi-wmi-platform ABBC0F6E-8EA1-11D1-00A0-C90629100000: WMI interface version 2.8
[ 2509.269356] msi-wmi-platform ABBC0F6E-8EA1-11D1-00A0-C90629100000: EC RAM version 0.1
[ 2509.269360] msi-wmi-platform ABBC0F6E-8EA1-11D1-00A0-C90629100000: EC firmware version 16R8IMS1.1070524202315:12:50
```

Wer-Wolf, 2024-02-15:
[msi-wmi-platform.zip](https://github.com/lm-sensors/lm-sensors/files/14302940/msi-wmi-platform.zip)

My bad, i wrongly initialized the input buffer. Does this one work?

Kalabaza, 2024-02-17:
With that new default value in the input buffer, now all the time it has these values:

```
msi_wmi_platform-virtual-0
Adapter: Virtual device
fan1:           FAULT
fan2:           FAULT
fan3:           FAULT
fan4:           FAULT
```

It only changes, when the fan is at the lowest speed and about to turn off:
```
fan1:        120000 RPM
fan1:        160000 RPM
```

Wer-Wolf, 2024-02-19:
[msi-wmi-platform.zip](https://github.com/lm-sensors/lm-sensors/files/14330063/msi-wmi-platform.zip)

An off-by-one error inside the index calculation is likely responsible for this, can you try again?

Kalabaza, 2024-02-21:
With that index fixed, now the data makes more sense:

Lowest speed:
`fan1:        1935 RPM`

Highest speed:
`fan1:        3609 RPM`

And going from the highest to the lowest is also fine now:
```
fan1:        3555 RPM
fan1:        3428 RPM
fan1:        3380 RPM
fan1:        3243 RPM
fan1:        3076 RPM
fan1:        2944 RPM
fan1:        2823 RPM
fan1:        2711 RPM
fan1:        2580 RPM
fan1:        2400 RPM
fan1:        2341 RPM
```

The only missing value is when the fan is off:
`fan1:           FAULT`

Kalabaza, 2024-02-21:
That is the value that I get:

`echo -n -e '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' > get_fan && hexedit get_fan`

`01 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00`

Wer-Wolf, 2024-02-21:
I assumed that only disabled fans have a speed value of `0x00`, which does not seem to be true.

I will publish an updated version tomorrow which drops the fan fault handling. This however means that you have to manually configure lm-sensors to ignore the non-existing fans.

glpnk, 2024-02-21:
Maybe fan count encoded in other WMI registers. Or hardcoded to MSI Center, because one of txt files inside has model name and fan quantity list

Wer-Wolf, 2024-02-21:
Should we find a way to query the WMI interface about the number of supported fans, then we still can extend the driver to support that.

For now i think hiding the non-existent fan via `sensors.conf` is sufficient.

Wer-Wolf, 2024-02-22:
[msi-wmi-platform.zip](https://github.com/lm-sensors/lm-sensors/files/14372892/msi-wmi-platform.zip)

This should be the final version of the driver. Can you verify that it works?

Kalabaza, 2024-02-29:
Hello and thank you for waiting for my answer, I just tried the module and is giving good results:

fan off
```
msi_wmi_platform-virtual-0
Adapter: Virtual device
fan1:           0 RPM
fan2:           0 RPM
fan3:           0 RPM
fan4:           0 RPM
```

fan on (slowest)
`fan1:        2341 RPM`

fan on (fastest)
`fan1:        3609 RPM`

The only strange thing is that when the fan is shutting down, it gives this speed:
`fan1:        480000 RPM`

Aside of that, the speeds are now fine and the non present fans have a value with zero.

Wer-Wolf, 2024-03-01:
Great to hear, i wonder whats behind this strange readings when the fan is shutting down.

Does the MSI software under windows have the same issue?

Wer-Wolf, 2024-03-01:
If the bogus fan speed is always the same and happens every time the fan is shutting down, then the driver can filter it out (there are likely no fans with 480000 RPM).

glpnk, 2024-03-02:
I thought that is related to timer overflow. But after the single cooler in the system jumped from Fan-1 position to Fan-3 I have no idea what the heck is going here. But my system is WMI1 and I reading data from EC with DD.

```
sudo dd if=/sys/kernel/debug/ec/ec0/io bs=1 skip=200 count=8 2>/dev/null | hexdump -C

[C8]    00 00 00 00 00 b2 14 00                           |........|
```

`CD: 0xB2` is some RPM value and `CE: 0x14` is some persistent garbage. `CD` goes to zero immediately after fan stops.

Other random ~~conspiracy~~ theory: Cooler is jittering after stop with short enough pulse time to be counted as `1 time point`

Wer-Wolf, 2024-03-02:
Ok, i will modify the driver to filter those values out.

Wer-Wolf, 2024-03-02:
[msi-wmi-platform.zip](https://github.com/lm-sensors/lm-sensors/files/14467858/msi-wmi-platform.zip)

Here it is, the bogus values should now be gone.

[46 more replies]