lm-sensors/lm-sensors #477: JSON output does not specify units/quantity

hramrach, 2023-12-23:
The -j option is not usable.

It's not specified what kind of measurement is provided by the sensor and what units it is in.

   "BAT1-acpi-0":{
      "Adapter": "ACPI interface",
      "in0":{
         "in0_input": 16.583
      },
      "curr1":{
         "curr1_input": 0.001
      }
   },
   "iwlwifi_1-virtual-0":{
      "Adapter": "Virtual device",
      "temp1":{
         "temp1_input": 43.000
      }
   },

WTF?

alexanderperlis, 2024-04-30:
Some information on measurement units appears in the [kernel hwmon documentation](https://www.kernel.org/doc/Documentation/hwmon/sysfs-interface); however, note that `sensors` re-scales some values (in`lm-sensors/lib/sysfs.c`, consult `get_type_scaling`). For example, a millidegrees Celsius value such as 16583 appearing in `/sys/class/hwmon` is output by `sensors` as 16.583 degrees Celsius.

The JSON could be improved in various ways. The proposal in [issues/494](494) would, among other things, include the measurement unit.

hramrach, 2024-05-01:
I implemented units in my fork, fwiw.

hramrach, 2024-05-01:
{
  "BAT1-isa-00ba": {
    "Adapter": "ISA adapter",
    "curr1": {
      "input": {
        "quantity": "current",
        "unit": "A",
        "value": 0
      }
    },
    "in0": {
      "input": {
        "quantity": "voltage",
        "unit": "V",
        "value": 16.575
      }
    }
  },
  "acpitz-isa-0000": {
    "Adapter": "ISA adapter",
    "temp1": {
      "crit": {
        "quantity": "temperature",
        "unit": "°C",
        "value": 210
      },
      "input": {
        "quantity": "temperature",
        "unit": "°C",
        "value": 43.8
      }
    },
    "temp2": {
      "crit": {
        "quantity": "temperature",
        "unit": "°C",
        "value": 210
      },
      "input": {
        "quantity": "temperature",
        "unit": "°C",
        "value": 42.8
      }
    },
    "temp3": {
      "crit": {
        "quantity": "temperature",
        "unit": "°C",
        "value": 210
      },
      "input": {
        "quantity": "temperature",
        "unit": "°C",
        "value": 43.8
      }
    },
    "temp4": {
      "crit": {
        "quantity": "temperature",
        "unit": "°C",
        "value": 210
      },
      "input": {
        "quantity": "temperature",
        "unit": "°C",
        "value": 39.8
      }
    },
    "temp5": {
      "crit": {
        "quantity": "temperature",
        "unit": "°C",
        "value": 210
      },
      "input": {
        "quantity": "temperature",
        "unit": "°C",
        "value": 29.8
      }
    }
  },
  "coretemp-isa-0000": {
    "Adapter": "ISA adapter",
    "temp1": {
      "crit": {
        "quantity": "temperature",
        "unit": "°C",
        "value": 100
      },
      "crit_alarm": {
        "quantity": "boolean",
        "value": 0
      },
      "input": {
        "quantity": "temperature",
        "unit": "°C",
        "value": 42
      },
      "label": "Package id 0",
      "max": {
        "quantity": "temperature",
        "unit": "°C",
        "value": 100
      }
    },