lm-sensors/lm-sensors #479: add T_SENSOR information for ASUS WD X299 SAGE/10G

zemerdon, 2024-01-04:
Product Name: WS X299 SAGE/10G
        Version: Rev 1.xx

Good morning,

I have a single thermistor attached to the T_SENSOR on the motherboard which shows up in the BIOS no problem.  Is there a way to get this to show up in lm-sensors or possibly provide information to help development to add this sensor ?

I have also added "GRUB_CMDLINE_LINUX="acpi_enforce_resources=lax" with no changes.

zemerdon, 2024-01-04:
[DSDT.dsl.gz](https://github.com/lm-sensors/lm-sensors/files/13836147/DSDT.dsl.gz)
DSDT.dsl here

Scoon1e, 2024-11-04:
Same here.

I use the T_Sensor to read the water temperature of my cooling system.
As mention by @zemerdon the sensor shows up in the BIOS. I also control the pump and fan speed via BIOS in dependency of the temperature measured by the T_Sensor.

On Windows I used HWMonitor and worked perfectly.