lm-sensors/lm-sensors #486: Intel Raptor Lake 13900K - Wrong CPU core numbers

SuhasSrinivasan, 2024-03-17:
Dear Developers,

Thank you for making this utility.

Currently the CPU core numbers displayed are incorrect.
It starts at 0 and increases by 4 until 32, after which it is contiguous until 47.
Please find below OS and CPU info.

```
Ubuntu 22.04.4 LTS
6.1.0-1035-oem
```

`cpu-info`
```
Packages:
	0: Intel 13th Gen Core i9-13900K
Microarchitectures:
	24x unknown
Cores:
	0: 2 processors (0-1), Intel unknown
	1: 2 processors (2-3), Intel unknown
	2: 2 processors (4-5), Intel unknown
	3: 2 processors (6-7), Intel unknown
	4: 2 processors (8-9), Intel unknown
	5: 2 processors (10-11), Intel unknown
	6: 2 processors (12-13), Intel unknown
	7: 2 processors (14-15), Intel unknown
	8: 1 processor (16), Intel unknown
	9: 1 processor (17), Intel unknown
	10: 1 processor (18), Intel unknown
	11: 1 processor (19), Intel unknown
	12: 1 processor (20), Intel unknown
	13: 1 processor (21), Intel unknown
	14: 1 processor (22), Intel unknown
	15: 1 processor (23), Intel unknown
	16: 1 processor (24), Intel unknown
	17: 1 processor (25), Intel unknown
	18: 1 processor (26), Intel unknown
	19: 1 processor (27), Intel unknown
	20: 1 processor (28), Intel unknown
	21: 1 processor (29), Intel unknown
	22: 1 processor (30), Intel unknown
	23: 1 processor (31), Intel unknown
Logical processors (System ID):
	0 (0): APIC ID 0x00000000
	1 (1): APIC ID 0x00000001
	2 (2): APIC ID 0x00000008
	3 (3): APIC ID 0x00000009
	4 (4): APIC ID 0x00000010
	5 (5): APIC ID 0x00000011
	6 (6): APIC ID 0x00000018
	7 (7): APIC ID 0x00000019
	8 (8): APIC ID 0x00000020
	9 (9): APIC ID 0x00000021
	10 (10): APIC ID 0x00000028
	11 (11): APIC ID 0x00000029
	12 (12): APIC ID 0x00000030
	13 (13): APIC ID 0x00000031
	14 (14): APIC ID 0x00000038
	15 (15): APIC ID 0x00000039
	16 (16): APIC ID 0x00000040
	17 (17): APIC ID 0x00000042
	18 (18): APIC ID 0x00000044
	19 (19): APIC ID 0x00000046
	20 (20): APIC ID 0x00000048
	21 (21): APIC ID 0x0000004a
	22 (22): APIC ID 0x0000004c
	23 (23): APIC ID 0x0000004e
	24 (24): APIC ID 0x00000050
	25 (25): APIC ID 0x00000052
	26 (26): APIC ID 0x00000054
	27 (27): APIC ID 0x00000056
	28 (28): APIC ID 0x00000058
	29 (29): APIC ID 0x0000005a
	30 (30): APIC ID 0x0000005c
	31 (31): APIC ID 0x0000005e
```

`sensors`
```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +31.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +26.0°C  (high = +80.0°C, crit = +100.0°C)
Core 4:        +25.0°C  (high = +80.0°C, crit = +100.0°C)
Core 8:        +26.0°C  (high = +80.0°C, crit = +100.0°C)
Core 12:       +24.0°C  (high = +80.0°C, crit = +100.0°C)
Core 16:       +25.0°C  (high = +80.0°C, crit = +100.0°C)
Core 20:       +26.0°C  (high = +80.0°C, crit = +100.0°C)
Core 24:       +31.0°C  (high = +80.0°C, crit = +100.0°C)
Core 28:       +24.0°C  (high = +80.0°C, crit = +100.0°C)
Core 32:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 33:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 34:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 35:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 36:       +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 37:       +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 38:       +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 39:       +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 40:       +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 41:       +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 42:       +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 43:       +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 44:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 45:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 46:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 47:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
```

Wer-Wolf, 2024-03-20:
The linux kernel cpu topology code has problems with modern intel cpus. Can you try a more recent kernel version and check if those issue still exists?

SuhasSrinivasan, 2024-03-26:
Hi, thank you for reviewing this issue.
`6.1.0-1035-oem` is the latest kernel available for this workstation and it is a production machine.

Lefuneste83, 2024-03-27:
Same issue here with i5 13500 on Proxmox.

Linux pve 6.5.13-3-pve #1 SMP PREEMPT_DYNAMIC PMX 6.5.13-3 (2024-03-20T10:45Z) x86_64 GNU/Linux

> cpu-info

Packages:
	0: Intel 13th Gen Core i5-13500
Microarchitectures:
	14x unknown
Cores:
	0: 2 processors (0-1), Intel unknown
	1: 2 processors (2-3), Intel unknown
	2: 2 processors (4-5), Intel unknown
	3: 2 processors (6-7), Intel unknown
	4: 2 processors (8-9), Intel unknown
	5: 2 processors (10-11), Intel unknown
	6: 1 processor (12), Intel unknown
	7: 1 processor (13), Intel unknown
	8: 1 processor (14), Intel unknown
	9: 1 processor (15), Intel unknown
	10: 1 processor (16), Intel unknown
	11: 1 processor (17), Intel unknown
	12: 1 processor (18), Intel unknown
	13: 1 processor (19), Intel unknown
Logical processors (System ID):
	0 (0): APIC ID 0x00000000
	1 (1): APIC ID 0x00000001
	2 (2): APIC ID 0x00000008
	3 (3): APIC ID 0x00000009
	4 (4): APIC ID 0x00000010
	5 (5): APIC ID 0x00000011
	6 (6): APIC ID 0x00000018
	7 (7): APIC ID 0x00000019
	8 (8): APIC ID 0x00000020
	9 (9): APIC ID 0x00000021
	10 (10): APIC ID 0x00000028
	11 (11): APIC ID 0x00000029
	12 (12): APIC ID 0x00000030
	13 (13): APIC ID 0x00000032
	14 (14): APIC ID 0x00000034
	15 (15): APIC ID 0x00000036
	16 (16): APIC ID 0x00000038
	17 (17): APIC ID 0x0000003a
	18 (18): APIC ID 0x0000003c
	19 (19): APIC ID 0x0000003e

> sensors

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +43.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +35.0°C  (high = +80.0°C, crit = +100.0°C)
Core 4:        +39.0°C  (high = +80.0°C, crit = +100.0°C)
Core 8:        +35.0°C  (high = +80.0°C, crit = +100.0°C)
Core 12:       +43.0°C  (high = +80.0°C, crit = +100.0°C)
Core 16:       +36.0°C  (high = +80.0°C, crit = +100.0°C)
Core 20:       +39.0°C  (high = +80.0°C, crit = +100.0°C)
Core 24:       +36.0°C  (high = +80.0°C, crit = +100.0°C)
Core 25:       +36.0°C  (high = +80.0°C, crit = +100.0°C)
Core 26:       +36.0°C  (high = +80.0°C, crit = +100.0°C)
Core 27:       +36.0°C  (high = +80.0°C, crit = +100.0°C)
Core 28:       +33.0°C  (high = +80.0°C, crit = +100.0°C)
Core 29:       +32.0°C  (high = +80.0°C, crit = +100.0°C)
Core 30:       +32.0°C  (high = +80.0°C, crit = +100.0°C)
Core 31:       +32.0°C  (high = +80.0°C, crit = +100.0°C)

Lefuneste83, 2024-03-27:
I don't know if this is related but I get the following warning at boot in dmesg :

[    8.156891] proc_thermal_pci 0000:00:04.0: enabling device (0000 -> 0002)
[    8.157123] proc_thermal_pci 0000:00:04.0: error: proc_thermal_add, will continue

Cannot find any useful info on the origin if this warning but it seems there is an issue getting some thermal probe information.

Chipset is B760. I cannot find any BIOS settings relating to this.

Wer-Wolf, 2024-03-27:
> I don't know if this is related but I get the following warning at boot in dmesg :
>
> [ 8.156891] proc_thermal_pci 0000:00:04.0: enabling device (0000 -> 0002) [ 8.157123] proc_thermal_pci [...]

This error should be harmless, maybe you can try if a BIOS update solves this.

Wer-Wolf, 2024-03-27:
Can someone of you check if kernel 6.8 fixes the core numbering issue?

xiaoxi404, 2024-04-23:
> Can someone of you check if kernel 6.8 fixes the core numbering issue?
I am using kernel 6.8.7, I have same problem.

`uname -r`
```
6.8.7-zen1-1-zen
```

`sensors`
```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +37.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +36.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +31.0°C  (high = +100.0°C, crit = +100.0°C)
Core 8:        +35.0°C  (high = +100.0°C, crit = +100.0°C)
Core 12:       +37.0°C  (high = +100.0°C, crit = +100.0°C)
Core 16:       +33.0°C  (high = +100.0°C, crit = +100.0°C)
Core 20:       +32.0°C  (high = +100.0°C, crit = +100.0°C)
Core 24:       +31.0°C  (high = +100.0°C, crit = +100.0°C)
Core 25:       +31.0°C  (high = +100.0°C, crit = +100.0°C)
Core 26:       +31.0°C  (high = +100.0°C, crit = +100.0°C)
Core 27:       +31.0°C  (high = +100.0°C, crit = +100.0°C)
Core 28:       +34.0°C  (high = +100.0°C, crit = +100.0°C)
Core 29:       +34.0°C  (high = +100.0°C, crit = +100.0°C)
Core 30:       +34.0°C  (high = +100.0°C, crit = +100.0°C)
Core 31:       +34.0°C  (high = +100.0°C, crit = +100.0°C)
```
`lscpu -e=MODELNAME,CORE,CPU`
```
MODELNAME                            CORE CPU
12th Gen Intel(R) Core(TM) i7-12700H    0   0
12th Gen Intel(R) Core(TM) i7-12700H    0   1
12th Gen Intel(R) Core(TM) i7-12700H    1   2
12th Gen Intel(R) Core(TM) i7-12700H    1   3
12th Gen Intel(R) Core(TM) i7-12700H    2   4
12th Gen Intel(R) Core(TM) i7-12700H    2   5
12th Gen Intel(R) Core(TM) i7-12700H    3   6
12th Gen Intel(R) Core(TM) i7-12700H    3   7
12th Gen Intel(R) Core(TM) i7-12700H    4   8
12th Gen Intel(R) Core(TM) i7-12700H    4   9
12th Gen Intel(R) Core(TM) i7-12700H    5  10
12th Gen Intel(R) Core(TM) i7-12700H    5  11
12th Gen Intel(R) Core(TM) i7-12700H    6  12
12th Gen Intel(R) Core(TM) i7-12700H    7  13
12th Gen Intel(R) Core(TM) i7-12700H    8  14
12th Gen Intel(R) Core(TM) i7-12700H    9  15
12th Gen Intel(R) Core(TM) i7-12700H   10  16
12th Gen Intel(R) Core(TM) i7-12700H   11  17
12th Gen Intel(R) Core(TM) i7-12700H   12  18
12th Gen Intel(R) Core(TM) i7-12700H   13  19
```

Wer-Wolf, 2024-04-25:
Kernel 6.9 should land a couple of fixes for the hyprid cpu topology parser, see [here](https://www.phoronix.com/news/Linux-6.9-APIC-x86-CPU-Topology) for details.

When 6.9 is released, i could check if this solves this issue, otherwise you might create a bug report on the linux kernel bugzilla.

ms178, 2024-06-22:
I am on Kernel 6.9.6 and with sensors, I still get:

```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +35.0°C  (high = +70.0°C, crit = +90.0°C)
Core 0:        +32.0°C  (high = +70.0°C, crit = +90.0°C)
Core 4:        +28.0°C  (high = +70.0°C, crit = +90.0°C)
Core 8:        +28.0°C  (high = +70.0°C, crit = +90.0°C)
Core 12:       +29.0°C  (high = +70.0°C, crit = +90.0°C)
Core 16:       +31.0°C  (high = +70.0°C, crit = +90.0°C)
Core 20:       +28.0°C  (high = +70.0°C, crit = +90.0°C)
Core 24:       +28.0°C  (high = +70.0°C, crit = +90.0°C)
Core 28:       +29.0°C  (high = +70.0°C, crit = +90.0°C)
Core 32:       +31.0°C  (high = +70.0°C, crit = +90.0°C)
Core 33:       +31.0°C  (high = +70.0°C, crit = +90.0°C)
Core 34:       +31.0°C  (high = +70.0°C, crit = +90.0°C)
Core 35:       +31.0°C  (high = +70.0°C, crit = +90.0°C)
Core 36:       +30.0°C  (high = +70.0°C, crit = +90.0°C)
Core 37:       +30.0°C  (high = +70.0°C, crit = +90.0°C)
Core 38:       +30.0°C  (high = +70.0°C, crit = +90.0°C)
Core 39:       +30.0°C  (high = +70.0°C, crit = +90.0°C)
Core 40:       +29.0°C  (high = +70.0°C, crit = +90.0°C)
Core 41:       +29.0°C  (high = +70.0°C, crit = +90.0°C)
Core 42:       +29.0°C  (high = +70.0°C, crit = +90.0°C)
Core 43:       +29.0°C  (high = +70.0°C, crit = +90.0°C)
```
```
❯ lscpu -e=MODELNAME,CORE,CPU
MODELNAME                    CORE CPU
Intel(R) Core(TM) i7-14700KF    0   0
Intel(R) Core(TM) i7-14700KF    0   1
Intel(R) Core(TM) i7-14700KF    1   2
Intel(R) Core(TM) i7-14700KF    1   3
Intel(R) Core(TM) i7-14700KF    2   4
Intel(R) Core(TM) i7-14700KF    2   5
Intel(R) Core(TM) i7-14700KF    3   6
Intel(R) Core(TM) i7-14700KF    3   7
Intel(R) Core(TM) i7-14700KF    4   8
Intel(R) Core(TM) i7-14700KF    4   9
Intel(R) Core(TM) i7-14700KF    5  10
Intel(R) Core(TM) i7-14700KF    5  11
Intel(R) Core(TM) i7-14700KF    6  12
Intel(R) Core(TM) i7-14700KF    6  13
Intel(R) Core(TM) i7-14700KF    7  14
Intel(R) Core(TM) i7-14700KF    7  15
Intel(R) Core(TM) i7-14700KF    8  16
Intel(R) Core(TM) i7-14700KF    9  17
Intel(R) Core(TM) i7-14700KF   10  18
Intel(R) Core(TM) i7-14700KF   11  19
Intel(R) Core(TM) i7-14700KF   12  20
Intel(R) Core(TM) i7-14700KF   13  21
Intel(R) Core(TM) i7-14700KF   14  22
Intel(R) Core(TM) i7-14700KF   15  23
Intel(R) Core(TM) i7-14700KF   16  24
Intel(R) Core(TM) i7-14700KF   17  25
Intel(R) Core(TM) i7-14700KF   18  26
Intel(R) Core(TM) i7-14700KF   19  27
```

Wer-Wolf, 2024-06-22:
Can you open an issue on the linux kernel bugzilla? When you do, i can try to notify the driver maintainer of you bugreport.

TheWyn, 2024-12-15:
Reporting this is also happening in kernel 6.13rc2 with KS variant as well

```
uname -r
6.13.0-rc2-1-cachyos-rc-rc2-5713-open-full
```

sensors

```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +35.0°C  (high = +95.0°C, crit = +115.0°C)
Core 0:        +28.0°C  (high = +95.0°C, crit = +115.0°C)
Core 4:        +32.0°C  (high = +95.0°C, crit = +115.0°C)
Core 8:        +30.0°C  (high = +95.0°C, crit = +115.0°C)
Core 12:       +30.0°C  (high = +95.0°C, crit = +115.0°C)
Core 16:       +33.0°C  (high = +95.0°C, crit = +115.0°C)
Core 20:       +32.0°C  (high = +95.0°C, crit = +115.0°C)
Core 24:       +30.0°C  (high = +95.0°C, crit = +115.0°C)
Core 28:       +31.0°C  (high = +95.0°C, crit = +115.0°C)
Core 32:       +30.0°C  (high = +95.0°C, crit = +115.0°C)
Core 33:       +30.0°C  (high = +95.0°C, crit = +115.0°C)
Core 34:       +30.0°C  (high = +95.0°C, crit = +115.0°C)
Core 35:       +30.0°C  (high = +95.0°C, crit = +115.0°C)
Core 36:       +29.0°C  (high = +95.0°C, crit = +115.0°C)
Core 37:       +29.0°C  (high = +95.0°C, crit = +115.0°C)
Core 38:       +29.0°C  (high = +95.0°C, crit = +115.0°C)
Core 39:       +29.0°C  (high = +95.0°C, crit = +115.0°C)
Core 40:       +28.0°C  (high = +95.0°C, crit = +115.0°C)
Core 41:       +28.0°C  (high = +95.0°C, crit = +115.0°C)
Core 42:       +28.0°C  (high = +95.0°C, crit = +115.0°C)
Core 43:       +28.0°C  (high = +95.0°C, crit = +115.0°C)
Core 44:       +26.0°C  (high = +95.0°C, crit = +115.0°C)
Core 45:       +24.0°C  (high = +95.0°C, crit = +115.0°C)
Core 46:       +24.0°C  (high = +95.0°C, crit = +115.0°C)
Core 47:       +24.0°C  (high = +95.0°C, crit = +115.0°C)
```

lscpu -e=MODELNAME,CORE,CPU

```
MODELNAME                             CORE CPU
13th Gen Intel(R) Core(TM) i9-13900KS    0   0
13th Gen Intel(R) Core(TM) i9-13900KS    0   1
13th Gen Intel(R) Core(TM) i9-13900KS    1   2
13th Gen Intel(R) Core(TM) i9-13900KS    1   3
13th Gen Intel(R) Core(TM) i9-13900KS    2   4
13th Gen Intel(R) Core(TM) i9-13900KS    2   5
13th Gen Intel(R) Core(TM) i9-13900KS    3   6
13th Gen Intel(R) Core(TM) i9-13900KS    3   7
13th Gen Intel(R) Core(TM) i9-13900KS    4   8
13th Gen Intel(R) Core(TM) i9-13900KS    4   9
13th Gen Intel(R) Core(TM) i9-13900KS    5  10
13th Gen Intel(R) Core(TM) i9-13900KS    5  11
13th Gen Intel(R) Core(TM) i9-13900KS    6  12
13th Gen Intel(R) Core(TM) i9-13900KS    6  13
13th Gen Intel(R) Core(TM) i9-13900KS    7  14
13th Gen Intel(R) Core(TM) i9-13900KS    7  15
13th Gen Intel(R) Core(TM) i9-13900KS    8  16
13th Gen Intel(R) Core(TM) i9-13900KS    9  17
13th Gen Intel(R) Core(TM) i9-13900KS   10  18
13th Gen Intel(R) Core(TM) i9-13900KS   11  19
13th Gen Intel(R) Core(TM) i9-13900KS   12  20
13th Gen Intel(R) Core(TM) i9-13900KS   13  21
13th Gen Intel(R) Core(TM) i9-13900KS   14  22
13th Gen Intel(R) Core(TM) i9-13900KS   15  23
13th Gen Intel(R) Core(TM) i9-13900KS   16  24
13th Gen Intel(R) Core(TM) i9-13900KS   17  25
13th Gen Intel(R) Core(TM) i9-13900KS   18  26
13th Gen Intel(R) Core(TM) i9-13900KS   19  27
13th Gen Intel(R) Core(TM) i9-13900KS   20  28
13th Gen Intel(R) Core(TM) i9-13900KS   21  29
13th Gen Intel(R) Core(TM) i9-13900KS   22  30
13th Gen Intel(R) Core(TM) i9-13900KS   23  31
```

LegendaryFire, 2025-02-02:
Has this issue ever been resolved? Experiencing the same issue after upgrading to a Intel i7 12700k from an Intel i3 12100.

Wer-Wolf, 2025-02-06:
I send the driver maintainer an email, hopefully he responds.