lm-sensors/lm-sensors #492: GIGABYTE G5 KF lm_sensors doesent detect Fans

CARBAJE, 2024-04-24:
Hello. I hope you're having a good day. My concern pertains to the scanning of some sensors during the execution of the command 'sensors-detect.' Specifically, the issue I'm encountering relates to the fans. In my case, it appears that 'sensors-detect' did not detect the fans sensors and others.

Here is the output of `sensors-detect`:

`
[carbaje@archlinux ~]$ sudo sensors-detect
[sudo] password for carbaje: 
# sensors-detect version 3.6.0+git
# System: GIGABYTE G5 KF5 [Not Applicable] (laptop)
# Kernel: 6.8.7-zen1-1-zen x86_64
# Processor: 13th Gen Intel(R) Core(TM) i7-13620H (6/186/2)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
AMD Family 19h thermal sensors...                           No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x5570
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 8086:51a3 at 0000:00:1f.4.
Sorry, no supported PCI bus adapters found.

Next adapter: SMBus I801 adapter at efa0 (i2c-0)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x48
Probing for `National Semiconductor LM75'...                No
Probing for `National Semiconductor LM75A'...               No
Probing for `Dallas Semiconductor DS75'...                  No
Probing for `National Semiconductor LM77'...                No
Probing for `Analog Devices ADT7410/ADT7420'...             No
Probing for `Analog Devices ADT7411'...                     No
Probing for `Maxim MAX6642'...                              No
Probing for `Texas Instruments TMP435'...                   No
Probing for `National Semiconductor LM73'...                No
Probing for `National Semiconductor LM92'...                No
Probing for `National Semiconductor LM76'...                No
Probing for `Maxim MAX6633/MAX6634/MAX6635'...              No
Probing for `NXP/Philips SA56004'...                        No
Probing for `SMSC EMC1023'...                               No
Probing for `SMSC EMC1043'...                               No
Probing for `SMSC EMC1053'...                               No
Probing for `SMSC EMC1063'...                               No
Client found at address 0x4a
Probing for `National Semiconductor LM75'...                No
Probing for `National Semiconductor LM75A'...               No
Probing for `Dallas Semiconductor DS75'...                  No
Probing for `National Semiconductor LM77'...                No
Probing for `Analog Devices ADT7410/ADT7420'...             No
Probing for `Analog Devices ADT7411'...                     No
Probing for `Maxim MAX6642'...                              No
Probing for `Texas Instruments TMP435'...                   No
Probing for `National Semiconductor LM73'...                No
Probing for `National Semiconductor LM92'...                No
Probing for `National Semiconductor LM76'...                No
Probing for `Maxim MAX6633/MAX6634/MAX6635'...              No
Probing for `NXP/Philips SA56004'...                        No
Client found at address 0x50
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No
Probing for `EDID EEPROM'...                                No
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No

Next adapter: Synopsys DesignWare I2C adapter (i2c-1)
Do you want to scan it? (YES/no/selectively): 
Adapter doesn't support all probing functions.
Some addresses won't be probed.

Next adapter: i915 gmbus dpa (i2c-2)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpb (i2c-3)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpc (i2c-4)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc1 (i2c-5)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc2 (i2c-6)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc3 (i2c-7)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc4 (i2c-8)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc5 (i2c-9)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc6 (i2c-10)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 1 at 1:00.0 (i2c-11)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 3 at 1:00.0 (i2c-12)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 5 at 1:00.0 (i2c-13)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AUX A/DDI A/PHY A (i2c-14)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AUX USBC4/DDI TC4/PHY TC4 (i2c-15)
Do you want to scan it? (yes/NO/selectively): 

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Do you want to overwrite /etc/conf.d/lm_sensors? (YES/no): 
[carbaje@archlinux ~]$ 

`

JLP, 2024-05-14:
I have the same `Trying family National Semiconductor/ITE... Yes Found unknown chip with ID 0x5570` output for KDE Slimbook V laptop (with AMD Ryzen 7 7840HS CPU). Looks like the chip in question is ITE IT8570E. I guess that is the one monitoring and controlling the fans and the rest. The `it87` hwmon kernel driver is currently missing support for this chip and it would be nice to add the support.

CARBAJE, 2024-05-20:
Thanks for the information,  I think it would be nice the support too