lm-sensors/lm-sensors #498: Can't detect any fan nor sensors on Pop_os

loddi-o, 2024-05-28:
OS: pop_os LTS

CONFIGURATION:
Mobo: [B550 a pro ](https://it.msi.com/Motherboard/B550-A-PRO) with 5x3pin (no PWM) case fans connected to individual system_fan headers, 3070ti FE (2 fans), cpu 5600X with an air cooler.

ISSUE:
On Windows I was able to control each of my fans just fine with Fan control. On Linux instead, I am having issues with detecting any fan. I installed fancontrol from pop shop, also [CoolerControl](https://gitlab.com/coolercontrol/coolercontrol), none of them detect any fan.

I tried with lm-sensors, executed `sudo sensors-detect` and answered YES to everything. At the end, I got this message:

```
lorenzo@pop-os:~$ sudo sensors-detect
# sensors-detect version 3.6.0
# System: Micro-Star International Co., Ltd. MS-7C56 [6.0]
# Board: Micro-Star International Co., Ltd. B550-A PRO (MS-7C56)
# Kernel: 6.8.0-76060800daily20240311-generic x86_64
# Processor: AMD Ryzen 5 5600X 6-Core Processor (25/33/2)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): YES
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           Success!
    (driver `k10temp')
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): YES
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd592

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): YES
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): YES
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): YES
Using driver `i2c-piix4' for device 0000:00:14.0: AMD KERNCZ SMBus

Next adapter: SMBus PIIX4 adapter port 0 at 0b00 (i2c-0)
Do you want to scan it? (YES/no/selectively): YES
Client found at address 0x4a
Probing for `National Semiconductor LM75'...                No
Probing for `National Semiconductor LM75A'...               No
Probing for `Dallas Semiconductor DS75'...                  No
Probing for `National Semiconductor LM77'...                No
Probing for `Analog Devices ADT7410/ADT7420'...             No
Probing for `Analog Devices ADT7411'...                     No
Probing for `Maxim MAX6642'...                              No
Probing for `Texas Instruments TMP435'...                   No
Probing for `National Semiconductor LM73'...                No
Probing for `National Semiconductor LM92'...                No
Probing for `National Semiconductor LM76'...                No
Probing for `Maxim MAX6633/MAX6634/MAX6635'...              No
Probing for `NXP/Philips SA56004'...                        No
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)
Client found at address 0x53
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)

Next adapter: SMBus PIIX4 adapter port 2 at 0b00 (i2c-1)
Do you want to scan it? (YES/no/selectively): YES

Next adapter: SMBus PIIX4 adapter port 1 at 0b20 (i2c-2)
Do you want to scan it? (YES/no/selectively): YES

Next adapter: NVIDIA i2c adapter 2 at 2b:00.0 (i2c-3)
Do you want to scan it? (yes/NO/selectively): YES

Next adapter: NVIDIA i2c adapter 5 at 2b:00.0 (i2c-4)
Do you want to scan it? (yes/NO/selectively): YES
YE
Next adapter: NVIDIA i2c adapter 6 at 2b:00.0 (i2c-5)
Do you want to scan it? (yes/NO/selectively): YES
YE
Next adapter: NVIDIA i2c adapter 7 at 2b:00.0 (i2c-6)
Do you want to scan it? (yes/NO/selectively): YES
Client found at address 0x58
Probing for `Analog Devices ADT7462'...                     No
Probing for `Andigilog aSC7512'...                          No
Client found at address 0x5c
Probing for `Analog Devices ADT7462'...                     No
Probing for `SMSC EMC1072'...                               No
Probing for `SMSC EMC1073'...                               No
Probing for `SMSC EMC1074'...                               No

Next adapter: NVIDIA i2c adapter 8 at 2b:00.0 (i2c-7)
Do you want to scan it? (yes/NO/selectively): YES

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `k10temp' (autoloaded):
  * Chip `AMD Family 17h thermal sensors' (confidence: 9)

No modules to load, skipping modules configuration.

Unloading cpuid... OK

```
and executed `sensors` again, Unfortunatly it seems like it didn't detect any fans nor sensors, the output seems very poor:
```
lorenzo@pop-os:/etc$ sensors
nvme-pci-0100
Adapter: PCI adapter
Composite:    +38.9°C  (low  =  -5.2°C, high = +79.8°C)
                       (crit = +84.8°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +37.8°C  
Tccd1:        +34.0°C 
```
 

I dont know what else to try out. I want the fans to ramp up within the gpu temps, my bios can't do that.

Any ideas?

Wer-Wolf, 2024-05-29:
Can you check if manually loading the `nct6683` module gives you access to the fans?

closed 2024-06-01 by loddi-o

zedMar65, 2025-04-24:
How exactly was this solved?

loddi-o, 2025-04-24:
> How exactly was this solved?

By installing this: https://github.com/Fred78290/nct6687d

zedMar65, 2025-04-24:
Thank you, and wow  that was a quick reply to a year old issue