lm-sensors/lm-sensors #499: Found unknown chip with ID 0xd441 - Nuvoton NCT6686D

artemklevtsov, 2024-05-30:
Hi,

I use Arch Linux with the Ryzen 9 7900X and Asrock X670E Steel Legend and latest linux kernel (6.9.2-zen1-1-zen).

Motherboard sensors is missing.

In manual I found the motherboard chip model (NCT6686D). Sensors detect show the following message:

```bash
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd441
```

hardware info:

```bash
Machine:
  Type: Desktop Mobo: ASRock model: X670E Steel Legend
    serial: <superuser required> UEFI: American Megatrends LLC. v: 3.01
    date: 05/13/2024
CPU:
  Info: 12-core model: AMD Ryzen 9 7900X bits: 64 type: MT MCP cache:
    L2: 12 MiB
  Speed (MHz): avg: 2056 min/max: 545/5733 cores: 1: 5463 2: 4410 3: 545
    4: 545 5: 545 6: 5462 7: 3216 8: 3472 9: 545 10: 545 11: 545 12: 5463
    13: 3378 14: 545 15: 545 16: 545 17: 545 18: 545 19: 4934 20: 545 21: 5376
    22: 545 23: 545 24: 545

```

sensors-detect full output:

```bash
# sensors-detect version 3.6.0+git
# System: ASRock X670E Steel Legend [Default string]
# Kernel: 6.9.2-zen1-1-zen x86_64
# Processor: AMD Ryzen 9 7900X 12-Core Processor (25/97/2)

Running in automatic mode, default answers to all questions
are assumed.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
AMD Family 19h thermal sensors...                           No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
[... 110 lines cut ...]
Client found at address 0x51
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No
Client found at address 0x53
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No

Next adapter: SMBus PIIX4 adapter port 2 at 0b00 (i2c-16)
Do you want to scan it? (YES/no/selectively): 

Next adapter: SMBus PIIX4 adapter port 1 at 0b20 (i2c-17)
Do you want to scan it? (YES/no/selectively): 

Sorry, no sensors were detected.
Either your system has no sensors, or they are not supported, or
they are connected to an I2C or SMBus adapter that is not
supported. If you find out what chips are on your board, check
https://hwmon.wiki.kernel.org/device_support_status for driver status.
```

[image]

Motherboard manual: https://download.asrock.com/Manual/X670E%20Steel%20Legend.pdf

Wer-Wolf, 2024-06-03:
What happens if you manually load the `nct6683` driver?

chrisaga, 2024-09-09:
Hello. Same motherboard here.
I have no nct6687 (ubuntu 22.04 LTS Jammy) but nct6683 partially works when forced.

~~~

$ sudo modprobe nct6683 force=onsudo modprobe nct6683 force=on
~~~

`dmesg` gives:
~~~
[ 1833.000594] nct6683: Found NCT6686D or compatible chip at 0x2e:0xa20
[ 1833.001804] nct6683 nct6683.2592: NCT6686D EC firmware version 1.0 build 07/19/22
~~~

`sensors` gives:
~~~
nct6686-isa-0a20
Adapter: ISA adapter
VIN0:               1.06 V  (min =  +0.00 V, max =  +0.00 V)
VIN1:               1.01 V  (min =  +0.00 V, max =  +0.00 V)
VIN2:               1.01 V  (min =  +0.00 V, max =  +0.00 V)
VIN3:               1.63 V  (min =  +0.00 V, max =  +0.00 V)
VIN5:               1.02 V  (min =  +0.00 V, max =  +0.00 V)
VIN6:               1.10 V  (min =  +0.00 V, max =  +0.00 V)
VIN7:               1.10 V  (min =  +0.00 V, max =  +0.00 V)
VCC:                3.36 V  (min =  +0.00 V, max =  +0.00 V)
fan1:              592 RPM  (min =    0 RPM)
fan2:              603 RPM  (min =    0 RPM)
fan3:                0 RPM  (min =    0 RPM)
fan4:             2803 RPM  (min =    0 RPM)
fan5:              412 RPM  (min =    0 RPM)
fan6:              386 RPM  (min =    0 RPM)
Thermistor 16:     +34.5°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = thermistor
Thermistor 14:     +36.5°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = thermistor
Thermistor 15:     +36.5°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = thermistor
AMD TSI Addr 98h:  +44.0°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = AMD AMDSI
intrusion0:       OK
beep_enable:      disabled
~~~

Config is needed to get correct data.

artemklevtsov, 2024-09-09:
@chrisaga thank you. It works for me too on Arch Linux. What about labels?

chrisaga, 2024-09-09:
> @chrisaga thank you. It works for me too on Arch Linux. What about labels?
Well ... some default lables are strange. They can be configured by creating a file such as `/etc/sensors.d/ASRock-X670E-Steel-Legend`
Fans are quite easy and voltages are more tricky.
For now I have: 
~~~
# Supporting ASRock X670E Steel Legend mother board
# with nuvoton nct6686D chip and nct6686 kernel module
chip "nct6686-isa-0a20"

    # Fans
    label fan1 "CPU Fan 1"
    label fan2 "CPU Fan 2"
    label fan3 "Chassis/WP Fan 4"
    label fan4 "Chassis/WP Fan 1"
    label fan5 "Chassis/WP Fan 3"
    label fan6 "Chassis/WP Fan 2"

    # Voltages
    label in0 "VCore (in0)"
    label in1 "?(in1)"
    label in2 "?(in2)"
    label in3 "?(in3)"
    label in4 "?(in4)"
    label in5 "?(in5)"
    label in6 "?(in6)"
    label in7 "+3.3V (in7)"
~~~

`sensors` displays:
~~~
nct6686-isa-0a20
Adapter: ISA adapter
VCore (in0):      432.00 mV (min =  +0.00 V, max =  +0.00 V)
?(in1):             1.01 V  (min =  +0.00 V, max =  +0.00 V)
?(in2):             1.01 V  (min =  +0.00 V, max =  +0.00 V)
?(in3):             1.63 V  (min =  +0.00 V, max =  +0.00 V)
?(in4):             1.02 V  (min =  +0.00 V, max =  +0.00 V)
?(in5):             1.10 V  (min =  +0.00 V, max =  +0.00 V)
?(in6):             1.10 V  (min =  +0.00 V, max =  +0.00 V)
+3.3V (in7):        3.36 V  (min =  +0.00 V, max =  +0.00 V)
CPU Fan 1:         604 RPM  (min =    0 RPM)
CPU Fan 2:         603 RPM  (min =    0 RPM)
Chassis/WP Fan 4:    0 RPM  (min =    0 RPM)
Chassis/WP Fan 1: 2777 RPM  (min =    0 RPM)
Chassis/WP Fan 3:  405 RPM  (min =    0 RPM)
Chassis/WP Fan 2:  389 RPM  (min =    0 RPM)
Thermistor 16:     +34.5°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = thermistor
Thermistor 14:     +36.5°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = thermistor
Thermistor 15:     +37.0°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = thermistor
AMD TSI Addr 98h:  +45.0°C  (low  =  +0.0°C)
                            (high =  +0.0°C, hyst =  +0.0°C)
                            (crit =  +0.0°C)  sensor = AMD AMDSI
intrusion0:       OK
beep_enable:      disabled
~~~
VCore and +3.3V are easy to guess because they don't need computation.
I tried to set min an max values but as far as I understand, nct6686 module doesn't allow that.

chrisaga, 2024-09-10:
@artemklevtsov I am getting some clues to decode the other voltages.

* in4 is VDDCR_SOC
* in1 and in2 both could be +5V or +12V
* in5 and in6 could be DRAM voltage
* in3 : don't know

Could you please post the result of `sensors -u` and a picture of the voltages displayed on your BIOS H/W monitor page ? Maybe we could get further.

artemklevtsov, 2024-09-10:
`sensors -u` output:

```bash
nct6686-isa-0a20
Adapter: ISA adapter
VCore:
  in0_input: 1.312
  in0_min: 0.000
  in0_max: 0.000
?(in1):
  in1_input: 0.992
  in1_min: 0.000
  in1_max: 0.000
?(in2):
  in2_input: 0.992
  in2_min: 0.000
  in2_max: 0.000
?(in3):
  in3_input: 1.696
  in3_min: 0.000
  in3_max: 0.000
?(in4):
  in4_input: 1.264
  in4_min: 0.000
  in4_max: 0.000
?(in5):
  in5_input: 1.120
  in5_min: 0.000
  in5_max: 0.000
?(in6):
  in6_input: 1.408
  in6_min: 0.000
  in6_max: 0.000
+3.3V:
  in7_input: 3.328
  in7_min: 0.000
  in7_max: 0.000
CPU Fan 1:
  fan1_input: 556.000
  fan1_min: 0.000
  fan1_pulses: 2.000
CPU Fan 2:
  fan2_input: 0.000
  fan2_min: 0.000
  fan2_pulses: 2.000
Chassis/WP Fan 4:
  fan3_input: 0.000
  fan3_min: 0.000
  fan3_pulses: 2.000
Chassis/WP Fan 1:
  fan4_input: 488.000
  fan4_min: 0.000
  fan4_pulses: 2.000
Chassis/WP Fan 3:
  fan5_input: 0.000
  fan5_min: 0.000
  fan5_pulses: 2.000
Chassis/WP Fan 2:
  fan6_input: 0.000
  fan6_min: 0.000
  fan6_pulses: 2.000
Thermistor 16:
  temp1_input: 43.000
  temp1_max: 0.000
  temp1_max_hyst: 0.000
  temp1_min: 0.000
  temp1_crit: 0.000
  temp1_type: 4.000
Thermistor 14:
  temp2_input: 48.000
  temp2_max: 0.000
  temp2_max_hyst: 0.000
  temp2_min: 0.000
  temp2_crit: 0.000
  temp2_type: 4.000
Thermistor 15:
  temp3_input: 45.000
  temp3_max: 0.000
  temp3_max_hyst: 0.000
  temp3_min: 0.000
  temp3_crit: 0.000
  temp3_type: 4.000
AMD TSI Addr 98h:
  temp4_input: 55.000
  temp4_max: 0.000
  temp4_max_hyst: 0.000
  temp4_min: 0.000
  temp4_crit: 0.000
  temp4_type: 5.000
intrusion0:
  intrusion0_alarm: 0.000
beep_enable:
  beep_enable: 0.000

```

artemklevtsov, 2024-09-10:
`sensors -u` output:

```bash
nct6686-isa-0a20
Adapter: ISA adapter
VCore:
  in0_input: 1.200
  in0_min: 0.000
  in0_max: 0.000
?(in1):
  in1_input: 0.992
  in1_min: 0.000
  in1_max: 0.000
?(in2):
  in2_input: 0.992
  in2_min: 0.000
  in2_max: 0.000
?(in3):
  in3_input: 1.696
  in3_min: 0.000
  in3_max: 0.000
?(in4):
  in4_input: 1.264
  in4_min: 0.000
  in4_max: 0.000
?(in5):
  in5_input: 1.120
  in5_min: 0.000
  in5_max: 0.000
?(in6):
  in6_input: 1.408
  in6_min: 0.000
  in6_max: 0.000
+3.3V:
  in7_input: 3.328
  in7_min: 0.000
  in7_max: 0.000
CPU Fan 1:
  fan1_input: 559.000
  fan1_min: 0.000
  fan1_pulses: 2.000
CPU Fan 2:
  fan2_input: 0.000
  fan2_min: 0.000
  fan2_pulses: 2.000
Chassis/WP Fan 4:
  fan3_input: 0.000
  fan3_min: 0.000
  fan3_pulses: 2.000
Chassis/WP Fan 1:
  fan4_input: 458.000
  fan4_min: 0.000
  fan4_pulses: 2.000
Chassis/WP Fan 3:
  fan5_input: 0.000
  fan5_min: 0.000
  fan5_pulses: 2.000
Chassis/WP Fan 2:
  fan6_input: 0.000
  fan6_min: 0.000
  fan6_pulses: 2.000
Thermistor 16:
  temp1_input: 42.000
  temp1_max: 0.000
  temp1_max_hyst: 0.000
  temp1_min: 0.000
  temp1_crit: 0.000
  temp1_type: 4.000
Thermistor 14:
  temp2_input: 46.500
  temp2_max: 0.000
  temp2_max_hyst: 0.000
  temp2_min: 0.000
  temp2_crit: 0.000
  temp2_type: 4.000
Thermistor 15:
  temp3_input: 44.000
  temp3_max: 0.000
  temp3_max_hyst: 0.000
  temp3_min: 0.000
  temp3_crit: 0.000
  temp3_type: 4.000
AMD TSI Addr 98h:
  temp4_input: 55.000
  temp4_max: 0.000
  temp4_max_hyst: 0.000
  temp4_min: 0.000
  temp4_crit: 0.000
  temp4_type: 5.000
intrusion0:
  intrusion0_alarm: 0.000
beep_enable:
  beep_enable: 0.000
```
BIOS picture: 
[image]

artemklevtsov, 2024-09-10:
I can repeat `sesnsors -u` with stress test (singe or multi-core). Will it help?

chrisaga, 2024-09-10:
Nice idea.
I just did that and saw my in6 showing alternatively 1.104 and 1.120.
I bet you would see something like that with your in5.
It would tend to confirm my hypothesis.

Unfortunately nothing else showed up.

artemklevtsov, 2024-09-10:
My results:

<details>
<summary>without stress</summary>

```
nct6686-isa-0a20
Adapter: ISA adapter
VCore:
  in0_input: 1.216
  in0_min: 0.000
  in0_max: 0.000
?(in1):
  in1_input: 0.992
  in1_min: 0.000
  in1_max: 0.000
?(in2):
  in2_input: 0.992
  in2_min: 0.000
  in2_max: 0.000
?(in3):
  in3_input: 1.696
  in3_min: 0.000
  in3_max: 0.000
?(in4):
  in4_input: 1.264
  in4_min: 0.000
  in4_max: 0.000
?(in5):
  in5_input: 1.120
  in5_min: 0.000
  in5_max: 0.000
?(in6):
  in6_input: 1.392
  in6_min: 0.000
  in6_max: 0.000
+3.3V:
  in7_input: 3.328
  in7_min: 0.000
  in7_max: 0.000
CPU Fan 1:
  fan1_input: 594.000
  fan1_min: 0.000
  fan1_pulses: 2.000
CPU Fan 2:
  fan2_input: 0.000
  fan2_min: 0.000
  fan2_pulses: 2.000
Chassis/WP Fan 4:
  fan3_input: 0.000
  fan3_min: 0.000
  fan3_pulses: 2.000
Chassis/WP Fan 1:
  fan4_input: 510.000
  fan4_min: 0.000
  fan4_pulses: 2.000
Chassis/WP Fan 3:
  fan5_input: 0.000
  fan5_min: 0.000
  fan5_pulses: 2.000
Chassis/WP Fan 2:
  fan6_input: 0.000
  fan6_min: 0.000
  fan6_pulses: 2.000
Thermistor 16:
  temp1_input: 43.000
  temp1_max: 0.000
  temp1_max_hyst: 0.000
  temp1_min: 0.000
  temp1_crit: 0.000
  temp1_type: 4.000
Thermistor 14:
  temp2_input: 49.000
  temp2_max: 0.000
  temp2_max_hyst: 0.000
  temp2_min: 0.000
  temp2_crit: 0.000
  temp2_type: 4.000
Thermistor 15:
  temp3_input: 45.000
  temp3_max: 0.000
  temp3_max_hyst: 0.000
  temp3_min: 0.000
  temp3_crit: 0.000
  temp3_type: 4.000
AMD TSI Addr 98h:
  temp4_input: 56.000
  temp4_max: 0.000
  temp4_max_hyst: 0.000
  temp4_min: 0.000
  temp4_crit: 0.000
  temp4_type: 5.000
intrusion0:
  intrusion0_alarm: 0.000
beep_enable:
  beep_enable: 0.000
  ```
</details>

<details>
<summary>stress -c 1</summary>

```
nct6686-isa-0a20
Adapter: ISA adapter
VCore:
  in0_input: 1.296
  in0_min: 0.000
  in0_max: 0.000
?(in1):
  in1_input: 0.992
  in1_min: 0.000
  in1_max: 0.000
?(in2):
  in2_input: 0.992
  in2_min: 0.000
  in2_max: 0.000
?(in3):
  in3_input: 1.696
  in3_min: 0.000
  in3_max: 0.000
?(in4):
  in4_input: 1.264
  in4_min: 0.000
  in4_max: 0.000
?(in5):
  in5_input: 1.120
  in5_min: 0.000
  in5_max: 0.000
?(in6):
  in6_input: 1.392
  in6_min: 0.000
  in6_max: 0.000
+3.3V:
  in7_input: 3.328
  in7_min: 0.000
  in7_max: 0.000
CPU Fan 1:
  fan1_input: 854.000
  fan1_min: 0.000
  fan1_pulses: 2.000
CPU Fan 2:
  fan2_input: 0.000
  fan2_min: 0.000
  fan2_pulses: 2.000
Chassis/WP Fan 4:
  fan3_input: 0.000
  fan3_min: 0.000
  fan3_pulses: 2.000
Chassis/WP Fan 1:
  fan4_input: 488.000
  fan4_min: 0.000
  fan4_pulses: 2.000
Chassis/WP Fan 3:
  fan5_input: 0.000
  fan5_min: 0.000
  fan5_pulses: 2.000
Chassis/WP Fan 2:
  fan6_input: 0.000
  fan6_min: 0.000
  fan6_pulses: 2.000
Thermistor 16:
  temp1_input: 43.000
  temp1_max: 0.000
  temp1_max_hyst: 0.000
  temp1_min: 0.000
  temp1_crit: 0.000
  temp1_type: 4.000
Thermistor 14:
  temp2_input: 48.500
  temp2_max: 0.000
  temp2_max_hyst: 0.000
  temp2_min: 0.000
  temp2_crit: 0.000
  temp2_type: 4.000
Thermistor 15:
  temp3_input: 45.500
  temp3_max: 0.000
  temp3_max_hyst: 0.000
  temp3_min: 0.000
  temp3_crit: 0.000
  temp3_type: 4.000
AMD TSI Addr 98h:
  temp4_input: 71.000
  temp4_max: 0.000
  temp4_max_hyst: 0.000
  temp4_min: 0.000
  temp4_crit: 0.000
  temp4_type: 5.000
intrusion0:
  intrusion0_alarm: 0.000
beep_enable:
  beep_enable: 0.000
```
</details>

<details>
<summary>stress -c 24</summary>

```
nct6686-isa-0a20
Adapter: ISA adapter
VCore:
  in0_input: 1.152
  in0_min: 0.000
  in0_max: 0.000
?(in1):
  in1_input: 0.992
  in1_min: 0.000
  in1_max: 0.000
?(in2):
  in2_input: 0.992
  in2_min: 0.000
  in2_max: 0.000
?(in3):
  in3_input: 1.696
  in3_min: 0.000
  in3_max: 0.000
?(in4):
  in4_input: 1.264
  in4_min: 0.000
  in4_max: 0.000
?(in5):
  in5_input: 1.136
  in5_min: 0.000
  in5_max: 0.000
?(in6):
  in6_input: 1.392
  in6_min: 0.000
  in6_max: 0.000
+3.3V:
  in7_input: 3.328
  in7_min: 0.000
  in7_max: 0.000
CPU Fan 1:
  fan1_input: 1234.000
  fan1_min: 0.000
  fan1_pulses: 2.000
CPU Fan 2:
  fan2_input: 0.000
  fan2_min: 0.000
  fan2_pulses: 2.000
Chassis/WP Fan 4:
  fan3_input: 0.000
  fan3_min: 0.000
  fan3_pulses: 2.000
Chassis/WP Fan 1:
  fan4_input: 489.000
  fan4_min: 0.000
  fan4_pulses: 2.000
Chassis/WP Fan 3:
  fan5_input: 0.000
  fan5_min: 0.000
  fan5_pulses: 2.000
Chassis/WP Fan 2:
  fan6_input: 0.000
  fan6_min: 0.000
  fan6_pulses: 2.000
Thermistor 16:
  temp1_input: 47.000
  temp1_max: 0.000
  temp1_max_hyst: 0.000
  temp1_min: 0.000
  temp1_crit: 0.000
  temp1_type: 4.000
Thermistor 14:
  temp2_input: 48.500
  temp2_max: 0.000
  temp2_max_hyst: 0.000
  temp2_min: 0.000
  temp2_crit: 0.000
  temp2_type: 4.000
Thermistor 15:
  temp3_input: 58.500
  temp3_max: 0.000
  temp3_max_hyst: 0.000
  temp3_min: 0.000
  temp3_crit: 0.000
  temp3_type: 4.000
AMD TSI Addr 98h:
  temp4_input: 80.000
  temp4_max: 0.000
  temp4_max_hyst: 0.000
  temp4_min: 0.000
  temp4_crit: 0.000
  temp4_type: 5.000
intrusion0:
  intrusion0_alarm: 0.000
beep_enable:
  beep_enable: 0.000
```
</details>

chrisaga, 2024-09-11:
So your in6 is the same as DRAM Voltage reported by your BIOS now.

I think I would stick with this:

`/etc/sensors.d/ASRock-X670E-Steel-Legend`
~~~
# HK Chrisaga 2024-09-11
# Supporting ASRock X670E Steel Legend mother board
# with nuvoton NCT6686D chip and nct6686 kernel module
# See https://www.nuvoton.com/export/resource-files/NCT6686D_HW_Datasheet_V0_5.pdf

chip "nct6686-isa-0a20"

    # Fans
    label fan1 "CPU Fan 1"
    label fan2 "CPU Fan 2"
    label fan3 "Chassis/WP Fan 4"
    label fan4 "Chassis/WP Fan 1"
    label fan5 "Chassis/WP Fan 3"
    label fan6 "Chassis/WP Fan 2"

    # Voltages
    # NCT6686D gives voltages values in range 0 to 2.048 V by steps of 16 mV (128 x 16 mV = 2.048 V)
    label in0 "VCore"
    #in1 and in2 read 1.008 x5 = 5.040 V :-)
    #                 1.008 x12 = 12.096 V :-)
    #Let's take a guess
    label in1 "+5V"
    compute in1 @*5, @/5
    label in2 "+12V"
    compute in2 @*12, @/12
    #in3 reads 1.632 -> x2 = 3.264 + 0.096 = 3.360
    # 96 mV / 2 = 48 mV = 3 x 16 mV
    # it could be some 3.3 V
    # could as well be a direct mesure 1.632 V = 102 x 16 mV
    # You can ignore it
    # ignore in3
    label in4 "VDDCR_SOC"
    label in5 "VDD_MISC"
    label in6 "DRAM"
    label in7 "+3.3V"
~~~

`/etc/modprobe.d/ASRock-x670e-steel-legend.conf`
~~~
# HK Chrisaga 2023-02-04
# force load superio module since NCT6686D with ID 0xd441 is not detected
options nct6683 force=1
~~~

`/etc/modules-load.d/ASRock-x670e-steel-legend.conf`
~~~
# Load module fore sensors
nct6683
~~~

cavity-dump, 2024-12-31:
No need to guess.
You can install the Asrock motherboard utility for windows. (the install works on wine too even if the program won't run)
Then in Program Files (x86)/ASROCK utility/A-Tuning/_conf you will find settings for all Asrock motherboards.
in this case you'll look in the X670ESTLLGD.xml file where you'll see:
..
        <Item Name="P5P0_V" Paging="0x01" Index="0x0A" Multiplier1="16" R1="0x192" R2="0x64" Divisor="1000"/>
        <Item Name="P12_V"  Paging="0x01" Index="0x0C" Multiplier1="16" R1="0X6E" R2="0X0A" Divisor="1000"/>           
..
so the 5V input uses 2 resistors R1=402, R2=100 => compute multiplier :  (R1/R2+1) = 5.02
and the 12V input uses 2 resistors R1=110, R2=10 => compute multiplier :  (R1/R2+1) = 12

compute in1 @*5.02, @/5.02
compute in2 @*12, @/12

You can also get the names / locations of all voltages if you look at the Index variable. You'll need to divide it by 2 and subtract 4: (each input takes 2 byte registers and the first 4 inputs are temperatures)
P5P0_V : 0x0A => 1
P12_V : 0x0C => 2

Now these settings are exactly the same for my Asrock motherboard even though it is a different one, so it wouldn't surprise me if many are the same.

cavity-dump, 2024-12-31:
Sorry, this was somehow purged
```
..
        <Item Name="P3P3_V" Paging="0x01" Index="0x16" Multiplier1="16" R1="0x0" R2="0x1" Divisor="1000"/>
        <Item Name="P5P0_V" Paging="0x01" Index="0x0A" Multiplier1="16" R1="0x192" R2="0x64" Divisor="1000"/>
..
```

jwolz, 2025-12-26:
In case someone needs, here is a config for Z890 Lightning WiFi (uses same XML as Z890 Steel Legend in the ASRock utility).
I hope I did not mix up some offsets from the XML (for the fans I only use CPU FAN1 and Chassis FAN1, so I can't verify the other values).

It works for me when I use force=1.

```
chip "nct6686-isa-0a20"
    # Voltages
    label in0 "CPU Vcore"

    #compute in1 @*5.02, @/5.02
    ignore in1
    
    label in2 "+5.0V"
    compute in2 @*5.02, @/5.02
    
    label in3 "+12V"
    compute in3 @*12, @/12
    
    label in4 "ATX+5VSB"
    compute in4 @*3.32, @/3.32
    
    label in5 "VCCSA"
    label in6 "VDD_CPU"
    label in7 "+3.3V"
    
    ignore in8
    
    
    # Temperature sensors
    label temp1 "CPU"
    label temp2 "M/B"
    label temp3 "MOS"
    # +20C is a guess, gives plausible values
    compute temp3 @+20, @-20
    label temp4 "T_SENSOR1"

    
    # Fan sensors
    label fan1 "CPU Fan1"
    label fan3 "CPU Fan2"
    label fan4 "Chassis Fan1"
    label fan5 "Chassis Fan2"
    label fan2 "Chassis Fan3"
    label fan6 "Chassis Fan4"
    label fan7 "Chassis Fan5"
    label fan8 "AIO_PUMP Fan"
    
```

SHADOW-XIII, 2026-05-09:
after spending some time on trying to be able to control the fans, I could not find a workable solution (please someone correct me if I am wrong)

with nct6686d (0xd441) you can only get sensors in read-only mode using nct6683 (as described above)

you won't be able to access and control pwm/fans speeds (pwmX_enable is not exposed or accessible)