lm-sensors/lm-sensors #502: Is `lm-sensors` completely useless - or am I doing something wrong?

seamusdemora, 2024-06-07:
I'm doing some research on the `hwmon` "SUBSYSTEM" in Linux. I started with the search term ['linux hwmon subsystem'](https://duckduckgo.com/?t=ffab&q=linux+hwmon+subsystem&ia=web). The very first item in the list from that search was (is) the [HWMon Wiki](https://hwmon.wiki.kernel.org/). In the FAQ section of that document, under the heading `Installation and Management`, it says: 

> Here are two simple guidelines:
>
>    Run sensors-detect and do what it tells you.
>    Always use modprobe, not insmod.

My system didn't have `sensors-detect`, and I then learned that it's part of `lm-sensors`. So, I `apt install'd` the `lm-sensors` package. Upon running `sensors-detect` I received the following (shown below) output/Q&A. And FWIW, **`sensors-detect` is also quite inaccurate**: I have the following `hwmon` devices on my system:
```bash
$ ls -l /sys/class/hwmon
total 0
lrwxrwxrwx 1 root root 0 Jun  7 20:39 hwmon0 -> ../../devices/virtual/thermal/thermal_zone0/hwmon0
lrwxrwxrwx 1 root root 0 Jun  7 20:39 hwmon1 -> ../../devices/platform/soc/3f804000.i2c/i2c-1/1-0068/hwmon/hwmon1
lrwxrwxrwx 1 root root 0 Jun  7 20:39 hwmon2 -> ../../devices/platform/soc/soc:firmware/raspberrypi-hwmon/hwmon/hwmon2
lrwxrwxrwx 1 root root 0 Jun  7 20:39 hwmon3 -> ../../devices/platform/soc/3f205000.i2c/i2c-0/0-0044/hwmon/hwmon3
```

I laughed out loud... what a completely and totally worthless exercise this was for me. I get that you guys are focused on PC-class hardware, but was it too painful to disclose that `lm-sensors` (and `sensors-detect` in particular) is useless on most/all other hardware platforms??? 

PLEASE do every non-PC-using-user of Linux a HUGE favor, and disclose the facts on this software! 

---
### Here lies the output of the wastrel `sensors-detect`: 

```
sudo sensors-detect
# sensors-detect version 3.6.0
# Kernel: 6.6.31+rpt-rpi-v7 armv7l
# Processor: ARMv7 Processor rev 4 (v7l)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): YES
modprobe: FATAL: Module cpuid not found in directory /lib/modules/6.6.31+rpt-rpi-v7
Failed to load module cpuid.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): YES
Sorry, no supported PCI bus adapters found.

Next adapter: bcm2835 (i2c@7e205000) (i2c-0)
Do you want to scan it? (YES/no/selectively): YES

Next adapter: bcm2835 (i2c@7e804000) (i2c-1)
Do you want to scan it? (YES/no/selectively): YES

Next adapter: bcm2835 (i2c@7e805000) (i2c-2)
Do you want to scan it? (YES/no/selectively): YES

Sorry, no sensors were detected.
Either your system has no sensors, or they are not supported, or
they are connected to an I2C or SMBus adapter that is not
supported. If you find out what chips are on your board, check
https://hwmon.wiki.kernel.org/device_support_status for driver status.
```

kkocdko, 2024-07-07:
Could you try this? I made an util for hwmon. Just `chmod +x ./hwmoncat ; ./hwmoncat`.

https://github.com/kkocdko/utils4linux/tree/master/hwmoncat

seamusdemora, 2024-08-14:
@kkocdko 

Hello - my apologies, I didn't intend to ignore your post, but I managed to resolve the issue in [another way](https://github.com/seamusdemora/PiFormulae/blob/master/Is_udev_brain-damaged.md). You can ignore the opening rant... I had to vent after the shitty experience with hwmon & lm-sensors.