lm-sensors/lm-sensors #504: RRD stops logging randomly

mjsir911, 2024-06-21:
Hi,

I'm running sensord with a 1 second interval rrd db started with this command:

```
/usr/sbin/sensord --rrd-file=/var/log/sensord.rrd --rrd-interval=1s --load-average --pid-file /run/sensord.pid
```

I'm not sure how to run in foreground or with debug mode, but it's running successfully for like an hour and then it just stops until a reboot.

Any ideas?

mjsir911, 2024-08-17:
syslog gets filled up with this:
```
 # tail /var/log/syslog
Aug 16 21:06:00 Ne0Ridl3y sensord: Error updating RRD file: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/senso
Aug 16 21:06:00 Ne0Ridl3y sensord: rrd update error (-1)
Aug 16 21:06:01 Ne0Ridl3y sensord: Error updating RRD file: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/senso
Aug 16 21:06:01 Ne0Ridl3y sensord: rrd update error (-1)
Aug 16 21:06:02 Ne0Ridl3y sensord: Error updating RRD file: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/se [... cut ...] d.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/sensord.rrd: /var/log/senso
Aug 16 21:06:04 Ne0Ridl3y sensord: rrd update error (-1)
```

mjsir911, 2024-10-19:
From the beginning of the log (I've added `rrd_update argv as well):

```
sensord: Error updating RRD file: '"sensord" "/var/log/sensord.rrd" "N:97.0:83.0:97.0:88.0:93.0:13.90:51.0:50.9:84.8:2.810000"' /var/log/sensord.rrd: could not lock RRD
```

mjsir911, 2024-11-13:
Alright I'm able to successfully reproduce by inducing a lock on the database with:

```
lockrun -L /var/log/sensord.rrd  -- sleep 1
```

Other important things to note, I've configured sensord to log out the sensor data every 1 second with `--rrd-interval=1s`, full invocation:

```
/usr/sbin/sensord --rrd-file=/var/log/sensord.rrd --rrd-interval=1s --load-average --pid-file /run/sensord.pid
```

kreddkrikk, 2025-01-10:
I was occasionally getting the same error when loading the web server graphs. I managed to fix it by setting the environment variable `RRD_LOCKING` to `block` in the sensord configuration file:

`RRD_LOCKING=block`

From the `rrdtool` manpage:

```
RRD_LOCKING
   If this environment variable is set, the RRD file is locked in the given mode:

   -   try fails, when the file is locked by another process.  This is the default.

   -   block waits until the lock is released.

   -   none skips locking at all.  Caller has to ensure a proper locking which should be compatible with "fcntl(fd, F_SETLK, ...)".

   Some utilities have command line options (--locking) which override this variable.
```