lm-sensors/lm-sensors #506: Missing fan information on HP EliteBook 860 16 inch G10 Notebook PC (96Z01ET)

rijnhard, 2024-08-01:
Output from sensors-detect (note: secure boot is on)

```
# sensors-detect version 3.6.0
# System: HP HP EliteBook 860 16 inch G10 Notebook PC [SBKPF] (laptop)
# Board: HP 8B41
# Kernel: 6.8.5-301.fc40.x86_64 x86_64
# Processor: 13th Gen Intel(R) Core(TM) i7-1365U (6/186/3)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 19h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
/dev/port: Operation not permitted

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
/dev/port: Operation not permitted

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 8086:51a3 at 0000:00:1f.4.
Sorry, no supported PCI bus adapters found.

Next adapter: Synopsys DesignWare I2C adapter (i2c-0)
Do you want to scan it? (YES/no/selectively): 
Adapter doesn't support all probing functions.
Some addresses won't be probed.
Client at address 0x2c can not be probed - unload all client drivers first!

Next adapter: i915 gmbus dpa (i2c-1)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpb (i2c-2)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpc (i2c-3)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc1 (i2c-4)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc2 (i2c-5)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc3 (i2c-6)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc4 (i2c-7)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc5 (i2c-8)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus tc6 (i2c-9)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AUX A/DDI A/PHY A (i2c-10)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AUX USBC1/DDI TC1/PHY TC1 (i2c-11)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AUX USBC2/DDI TC2/PHY TC2 (i2c-12)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AUX USBC3/DDI TC3/PHY TC3 (i2c-13)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AUX USBC4/DDI TC4/PHY TC4 (i2c-14)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: SMBus I801 adapter at efa0 (i2c-15)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x48
Probing for `National Semiconductor LM75'...                No
Probing for `National Semiconductor LM75A'...               No
Probing for `Dallas Semiconductor DS75'...                  No
Probing for `National Semiconductor LM77'...                No
Probing for `Analog Devices ADT7410/ADT7420'...             No
Probing for `Analog Devices ADT7411'...                     No
Probing for `Maxim MAX6642'...                              No
Probing for `Texas Instruments TMP435'...                   No
Probing for `National Semiconductor LM73'...                No
Probing for `National Semiconductor LM92'...                No
Probing for `National Semiconductor LM76'...                No
Probing for `Maxim MAX6633/MAX6634/MAX6635'...              No
Probing for `NXP/Philips SA56004'...                        No
Probing for `SMSC EMC1023'...                               No
Probing for `SMSC EMC1043'...                               No
Probing for `SMSC EMC1053'...                               No
Probing for `SMSC EMC1063'...                               No
Client found at address 0x50
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 No
Probing for `EDID EEPROM'...                                No

Next adapter: DPMST (i2c-16)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: DPMST (i2c-17)
Do you want to scan it? (yes/NO/selectively): 

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)
```

Output from sensors
```
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +41.0°C  

BAT0-acpi-0
Adapter: ACPI interface
in0:          13.07 V  

nvme-pci-0200
Adapter: PCI adapter
Composite:    +39.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +39.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +39.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +71.0°C  
temp2:        +25.0°C  
temp3:        +36.0°C  
temp4:        +46.0°C  
temp5:        +26.0°C  
temp6:        +45.0°C  
temp7:        +43.0°C  
temp8:         +0.0°C  
temp9:        +30.0°C  
temp10:       +30.0°C  
temp11:       +30.0°C  
temp12:       +30.0°C  
temp13:        +0.0°C  
```

According to the error in sensors-detect, this was not detected correctly:
```
lspci -nn | grep 1f.4
00:1f.4 SMBus [0c05]: Intel Corporation Alder Lake PCH-P SMBus Host Controller [8086:51a3] (rev 01)
```

ACPI Dump
[acpidump.txt](https://github.com/user-attachments/files/16457977/acpidump.txt)

Bmof
[05901221-D566-11D1-B2F0-00A0C9062910.zip](https://github.com/user-attachments/files/16458030/05901221-D566-11D1-B2F0-00A0C9062910.zip)

Report Descriptors
```
ls -l /sys/bus/hid/devices/00*
<snip>
/sys/bus/hid/devices/0018:04F3:31ED.0001 -> ../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-ELAN07AD:00/0018:04F3:31ED.0001
```
[0018:04F3:31ED.0001.txt](https://github.com/user-attachments/files/16458070/0018.04F3.31ED.0001.txt)

I followed the trail of what was needed in this report https://github.com/lm-sensors/lm-sensors/issues/475
It seems the devices are the same (from my uneducated eyes), if theres anything more I can provide, please let me know. I am dualbooting with windows 11 as well.

Wer-Wolf, 2024-08-13:
Sorry for my late response, i will take a look at the ACPI dump soon.

Wer-Wolf, 2024-08-13:
Does the `hp-wmi-sensors` driver work on you machine?

rijnhard, 2024-08-19:
> Does the `hp-wmi-sensors` driver work on you machine?

It appears not:
```
journalctl -b  | grep wmi
kernel: wmi_bus wmi_bus-PNP0C14:00: [Firmware Info]: DF4E63B6-3BBC-4858-9737-C74F82F821F3 has zero instances
kernel: wmi_bus wmi_bus-PNP0C14:00: [Firmware Info]: 8F1F6436-9F42-42C8-BADC-0E9424F20C9A has zero instances
kernel: wmi_bus wmi_bus-PNP0C14:02: [Firmware Bug]: WQZZ data block query control method not found
kernel: hp_wmi: query 0x4 returned error 0x5
```

Wer-Wolf, 2024-08-19:
Can you check if a BIOS update is available for your machine?

rijnhard, 2024-08-20:
Already up to date, it's supported by lvfs. I even had to grow my EFI
partition to make the update go through.

Wer-Wolf, 2024-08-20:
I think there exists another way to retrieve fan speed data on your machine, but the associated ACPI bytecode is hard to understand. I will give it a try, but this might take some time.

rijnhard, 2024-08-21:
OK I was wrong, `hp-wmi` very much is loaded, the kernel messages are just for specific elements not being supported, but those are fairly normal.

```shell
rijnhard@elite-rijn:~$ tree -pug /sys/class/hwmon/hwmon4
[lrwxrwxrwx root     root    ]  /sys/class/hwmon/hwmon4
├── [lrwxrwxrwx root     root    ]  device -> ../../../hp-wmi
├── [-r--r--r-- root     root    ]  name
├── [drwxr-xr-x root     root    ]  power
│   ├── [-rw-r--r-- root     root    ]  autosuspend_delay_ms
│   ├── [-rw-r--r-- root     root    ]  control
│   ├── [-r--r--r-- root     root    ]  runtime_active_time
│   ├── [-r--r--r-- root     root    ]  runtime_status
│   └── [-r--r--r-- root     root    ]  runtime_suspended_time
├── [-rw-r--r-- root     root    ]  pwm1_enable
├── [lrwxrwxrwx root     root    ]  subsystem -> ../../../../../class/hwmon
└── [-rw-r--r-- root     root    ]  uevent

4 directories, 8 files
rijnhard@elite-rijn:~$ sudo cat /sys/class/hwmon/hwmon4/name 
hp
rijnhard@elite-rijn:~$ sudo cat /sys/class/hwmon/hwmon4/pwm1_enable 
cat: /sys/class/hwmon/hwmon4/pwm1_enable: No data available
```

But it seems the fan information isn't reported, and something is funny with `pwm1_enable`, not sure if that's SecureBoot related though.

rijnhard, 2025-08-28:
@Wer-Wolf lets do it :) happy to test whatever you need, it might take me a few days but I'll happy figure out the bits to test whatever is needed.
Also prepped my fedora for custom kernels but haven't actually built one yet

Wer-Wolf, 2025-09-04:
[acpi-fan.zip](https://github.com/user-attachments/files/22142068/acpi-fan.zip)

Can test if the fan speed reporting works using the above kernel module? Please note that you need to unload the original ACPI fan driver using `sudo modprobe -r fan`.

Wer-Wolf, 2025-10-29:
The necessary patches are now upstream and will likely land with kernel 6.19. If fan speed reporting is still broken with the 6.19 kernel please send me a note.