lm-sensors/lm-sensors #508: Intel nuc x15 missing fan information

cyear, 2024-08-23:
`nian:~/ $ sudo sensors-detect                                    [11:13:02]
# sensors-detect version 3.6.0
# System: Intel(R) Client Systems LAPAC71H [M81843-405] (laptop)
# Board: Intel Corporation LAPAC71H
# Kernel: 6.10.6-gentoo-x86_64 x86_64
# Processor: 12th Gen Intel(R) Core(TM) i7-12700H (6/154/3)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Gentoo config file protection is active. Every file this program will
modify must be merged before the change will become active using
default Gentoo tools such as dispatch-conf, cfg-update, and etc-update.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Found unknown SMBus adapter 8086:51a3 at 0000:00:1f.4.
Sorry, no supported PCI bus adapters found.
Module i2c-dev loaded successfully.
[... 170 lines cut ...]
BAT0-acpi-0
Adapter: ACPI interface
in0:          16.59 V  
curr1:         0.00 A  

i915-pci-0300
Adapter: PCI adapter
in0:           0.00 V  
power1:           N/A  (max = 100.00 W)
energy1:      56.18 kJ

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +51.0°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +35.9°C  (low  = -273.1°C, high = +89.8°C)
                       (crit = +94.8°C)
Sensor 1:     +35.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +34.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +62.0°C  `

Wer-Wolf, 2024-08-25:
Can you share the output of `acpidump`?

cyear, 2024-08-25:
> 你能分享一下输出吗`acpidump`？

[dump.txt](https://github.com/user-attachments/files/16738549/dump.txt)

Wer-Wolf, 2024-08-25:
It seems that you machine does not support reading of fan speed data, sorry.

cyear, 2024-08-25:
> 看来您的机器不支持读取风扇速度数据，抱歉。

[image]
But on windows, you can only view it with the intel console

cyear, 2024-08-25:
> > 看来您的机器不支持读取风扇速度数据，抱歉。
> 
> ![屏幕截图 2024-08-25 221546](https://private-user-images.githubusercontent.com/77969791/361237758-49e6287b-7502-4acd-8fe5-dec515a447bb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ1OTU4MjcsIm5iZiI6MTcyNDU5NTUyNywicGF0aCI6Ii83Nzk2OTc5MS8zNjEyMzc3NTgtNDllNjI4N2ItNzUwMi00YWNkLThmZTUtZGVjNTE1YTQ0N2JiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODI1VDE0MTg0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM1N2FmZmFiYjU3Yjc1YjhjM2YwYWE0MzlhZTg3OGMyNDg0YTZmNWNmN2MwYmZhNGNhYTkyOGFhNTBkOGM0ODMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.LwZxyKEob6yv4VO6q2Zg3hF9sJYLa-7TqjUxliQm-Qo) But on windows, you can only view it with the intel console

How can I see what the software does to the device, what it writes, what it reads

Wer-Wolf, 2024-08-25:
I suspect it uses the CSID_WMI WMI class to communicate with the hardware.

The `WMAA` ACPI method is used for that, and intel also published some documentation for this interface.

However i have no idea how to read the fan speed.

cyear, 2024-08-28:
> I suspect it uses the CSID_WMI WMI class to communicate with the hardware.
> 
> The `WMAA` ACPI method is used for that, and intel also published some documentation for this interface.
> 
> However i have no idea how to read the fan speed.

Thanks for your reply. I found a project about nuc. After trying steps 1 and 2, I can check the fan speed ([https://github.com/MioNya4/fanctl](url)）

cyear, 2024-08-28:
> Thanks for your reply. I found a project about nuc. After trying steps 1 and 2, I can check the fan speed ([https://github.com/MioNya4/fanctl](url)）

[image]

cyear, 2024-08-28:
> ![image](https://private-user-images.githubusercontent.com/77969791/362326301-74efe27c-0be8-45d3-aab6-ba5dc0dba9b3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

But this doesn't seem to work completely
[image]

Wer-Wolf, 2024-08-28:
Cool, can you send me the output of `dmidecode`?

cyear, 2024-08-28:
> Cool, can you send me the output of `dmidecode`?

OK, thank you very much
[out.txt](https://github.com/user-attachments/files/16786876/out.txt)

Wer-Wolf, 2024-08-28:
I will take a look at the driver, can you try to update your BIOS?

cyear, 2024-08-28:
> I will take a look at the driver, can you try to update your BIOS?
Updated to the latest bios
[out.txt](https://github.com/user-attachments/files/16789919/out.txt)

Wer-Wolf, 2024-09-02:
I found another driver [here](https://github.com/tuxedocomputers/tuxedo-drivers).

I will try to sort of merge both drivers and create a stripped-down version which only supports the hwmon interface, but should be suitable for upstreaming.

cyear, 2024-09-05:
> I found another driver [here](https://github.com/tuxedocomputers/tuxedo-drivers).
> 
> I will try to sort of merge both drivers and create a stripped-down version which only supports the hwmon interface, but should be suitable for upstreaming.

Thank you very much

Wer-Wolf, 2024-09-09:
[uniwill-laptop.zip](https://github.com/user-attachments/files/16936581/uniwill-laptop.zip)

Can you compile this prototype driver an test if:
- the fan speed does not change by simply loading the module
- fan and temperature sensors work
- switching between manual and automatic fan control works
- suspend/resume correctly restores fan state

cyear, 2024-09-10:
> [uniwill-laptop.zip](https://github.com/user-attachments/files/16936581/uniwill-laptop.zip)
> 
> Can you compile this prototype driver and test if:
> 
> * the fan speed does not change by simply loading the module
> * fan and temperature sensors work
> * switching between manual and automatic fan control works
> * suspend/resume correctly restores fan state

```
nian:uniwill-laptop/ $ ls                                                                                                                         [9:05:48]
Makefile        modules.order     uniwill-laptop.ko   uniwill-laptop.mod.c  uniwill-laptop.o
Module.symvers  uniwill-laptop.c  uniwill-laptop.mod  uniwill-laptop.mod.o
```
What should I do next?

Wer-Wolf, 2024-09-10:
`sudo insmod unuwill-laptop.ko` should load the module.

cyear, 2024-09-10:
> `sudo insmod unuwill-laptop.ko`应该加载模块。

```
nian:uniwill-laptop/ $ lsmod |grep lapt                           [9:57:03]
uniwill_laptop         20480  0
wmi                    28672  4 video,uniwill_laptop,asus_wmi,wmi_bmof
nian:uniwill-laptop/ $ sensors                                    [9:57:13]
i915-pci-0300
Adapter: PCI adapter
in0:           0.00 V  
power1:           N/A  (max = 100.00 W)
energy1:      70.96 kJ

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +67.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 8:        +67.0°C  (high = +100.0°C, crit = +100.0°C)
Core 12:       +60.0°C  (high = +100.0°C, crit = +100.0°C)
Core 16:       +59.0°C  (high = +100.0°C, crit = +100.0°C)
Core 20:       +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 24:       +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 25:       +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 26:       +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 27:       +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 28:       +55.0°C  (high = +100.0°C, crit = +100.0°C)
Core 29:       +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 30:       +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 31:       +56.0°C  (high = +100.0°C, crit = +100.0°C)

BAT0-acpi-0
Adapter: ACPI interface
in0:          16.75 V  
curr1:         0.00 A  

ucsi_source_psy_USBC000:002-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           5.00 V  (min =  +5.00 V, max =  +5.00 V)
curr1:         0.00 A  (max =  +3.00 A)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +46.0°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +35.9°C  (low  = -273.1°C, high = +89.8°C)
                       (crit = +94.8°C)
Sensor 1:     +35.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +32.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +63.0°C  

nian:uniwill-laptop/ $                    
```
The qc71_laptop module was uninstalled by me and I am currently using your module

cyear, 2024-09-10:
> `sudo insmod unuwill-laptop.ko` should load the module.

```
[   72.741179] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: Project ID: 192
[   72.741187] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: probe with driver uniwill failed with error -5
```
This is from the dmesg output, maybe an error about the module?

Wer-Wolf, 2024-09-10:
Can you send me the full output of `dmesg` when loading the module?

Wer-Wolf, 2024-09-10:
[uniwill-laptop.zip](https://github.com/user-attachments/files/16942605/uniwill-laptop.zip)

It seems i forgot to allow reads to a crucial register, can you test this version?

cyear, 2024-09-10:
> [uniwill-laptop.zip](https://github.com/user-attachments/files/16942605/uniwill-laptop.zip)
> 
> It seems i forgot to allow reads to a crucial register, can you test this version?

```
[   12.858079] Bluetooth: RFCOMM ver 1.11
[   43.658684] warning: `ThreadPoolForeg' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[   49.874155] tun: Universal TUN/TAP device driver, 1.6
[   64.750937] i915 0000:00:02.0: Using 39-bit DMA addresses
[  100.832175] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: Project ID: 192
nian:uniwill-laptop/ $ sensors                                                                                                                   [17:31:24]
ucsi_source_psy_USBC000:002-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           5.00 V  (min =  +5.00 V, max =  +5.00 V)
curr1:         0.00 A  (max =  +3.00 A)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 8:        +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 12:       +53.0°C  (high = +100.0°C, crit = +100.0°C)
Core 16:       +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 20:       +50.0°C  (high = +100.0°C, crit = +100.0°C)
Core 24:       +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 25:       +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 26:       +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 27:       +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 28:       +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 29:       +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 30:       +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 31:       +51.0°C  (high = +100.0°C, crit = +100.0°C)

BAT0-acpi-0
Adapter: ACPI interface
in0:          16.74 V  
curr1:         0.00 A  

uniwill-virtual-0
Adapter: Virtual device
Main:        61564 RPM
Secondary:   61564 RPM
CPU:         +31984.0°C  
GPU:         +31984.0°C  

i915-pci-0300
Adapter: PCI adapter
in0:           0.00 V  
power1:           N/A  (max = 100.00 W)
energy1:       3.13 kJ

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +41.0°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +42.9°C  (low  = -273.1°C, high = +89.8°C)
                       (crit = +94.8°C)
Sensor 1:     +42.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +30.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +58.0°C  

nian:uniwill-laptop/ $      
```

The data seems to be abnormal

cyear, 2024-09-10:
> Can you send me the full output of `dmesg` when loading the module?加载模块时，您能否将 `dmesg` 的完整输出发送给我？

[dmesg.txt](https://github.com/user-attachments/files/16942713/dmesg.txt)

Wer-Wolf, 2024-09-10:
It seems that the prototype driver still needs some work. I will try fix those issue.

Wer-Wolf, 2024-09-10:
[uniwill-laptop.zip](https://github.com/user-attachments/files/16943131/uniwill-laptop.zip)

Does it work now?

cyear, 2024-09-10:
> [uniwill-laptop.zip](https://github.com/user-attachments/files/16943131/uniwill-laptop.zip)
> 
> Does it work now?

```
[ 2691.156782] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: Project ID: 192
nian:uniwill-laptop/ $ sensors                                                                                                                   [18:14:36]
ucsi_source_psy_USBC000:002-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           5.00 V  (min =  +5.00 V, max =  +5.00 V)
curr1:         0.00 A  (max =  +3.00 A)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +53.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +52.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 8:        +52.0°C  (high = +100.0°C, crit = +100.0°C)
Core 12:       +52.0°C  (high = +100.0°C, crit = +100.0°C)
Core 16:       +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 20:       +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 24:       +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 25:       +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 26:       +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 27:       +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 28:       +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 29:       +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 30:       +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 31:       +49.0°C  (high = +100.0°C, crit = +100.0°C)

BAT0-acpi-0
Adapter: ACPI interface
in0:          16.74 V  
curr1:         0.00 A  

uniwill-virtual-0
Adapter: Virtual device
Main:        61628 RPM
Secondary:   61628 RPM
CPU:         +48368.0°C  
GPU:         +48368.0°C  

i915-pci-0300
Adapter: PCI adapter
in0:           0.00 V  
power1:           N/A  (max = 100.00 W)
energy1:     264.79 kJ

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +35.0°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +35.9°C  (low  = -273.1°C, high = +89.8°C)
                       (crit = +94.8°C)
Sensor 1:     +35.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +25.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +54.0°C  

nian:uniwill-laptop/ $                  
```
can

Wer-Wolf, 2024-09-10:
[uniwill-laptop.zip](https://github.com/user-attachments/files/16943635/uniwill-laptop.zip)

Can you load this module and send the the output of dmesg? It should contain some debug data.

Wer-Wolf, 2024-09-10:
[uniwill-laptop.zip](https://github.com/user-attachments/files/16943736/uniwill-laptop.zip)

I think i found the issue, can you check this and report back?

cyear, 2024-09-10:
> [uniwill-laptop.zip](https://github.com/user-attachments/files/16943736/uniwill-laptop.zip)
> 
> I think i found the issue, can you check this and report back?我想我发现了问题，您能检查一下并报告吗？

```
[ 5763.198172] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1856 VALUE 8
[ 5763.198195] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: Project ID: 8
[ 5763.266973] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1893 VALUE 33792
[ 5763.338295] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1857 VALUE 512
nian:uniwill-laptop/ $ sensors                                                                                                                   [19:05:46]
ucsi_source_psy_USBC000:002-isa-0000
Adapter: ISA adapter
in0:           0.00 V  (min =  +0.00 V, max =  +0.00 V)
curr1:         0.00 A  (max =  +0.00 A)

ucsi_source_psy_USBC000:001-isa-0000
Adapter: ISA adapter
in0:           5.00 V  (min =  +5.00 V, max =  +5.00 V)
curr1:         0.00 A  (max =  +3.00 A)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +78.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +51.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +61.0°C  (high = +100.0°C, crit = +100.0°C)
Core 8:        +61.0°C  (high = +100.0°C, crit = +100.0°C)
Core 12:       +73.0°C  (high = +100.0°C, crit = +100.0°C)
Core 16:       +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 20:       +62.0°C  (high = +100.0°C, crit = +100.0°C)
Core 24:       +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 25:       +54.0°C  (high = +100.0°C, crit = +100.0°C)
Core 26:       +55.0°C  (high = +100.0°C, crit = +100.0°C)
Core 27:       +55.0°C  (high = +100.0°C, crit = +100.0°C)
Core 28:       +55.0°C  (high = +100.0°C, crit = +100.0°C)
Core 29:       +55.0°C  (high = +100.0°C, crit = +100.0°C)
Core 30:       +55.0°C  (high = +100.0°C, crit = +100.0°C)
Core 31:       +55.0°C  (high = +100.0°C, crit = +100.0°C)

BAT0-acpi-0
Adapter: ACPI interface
in0:          16.74 V  
curr1:         0.00 A  

uniwill-virtual-0
Adapter: Virtual device
Main:        1849 RPM
Secondary:   1855 RPM
CPU:          +72.0°C  
GPU:         +28966.0°C  

i915-pci-0300
Adapter: PCI adapter
in0:           0.00 V  
power1:           N/A  (max = 100.00 W)
energy1:     526.28 kJ

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +34.0°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +35.9°C  (low  = -273.1°C, high = +89.8°C)
                       (crit = +94.8°C)
Sensor 1:     +35.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +25.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +72.0°C  

nian:uniwill-laptop/ $      
```
Still not normal

cyear, 2024-09-10:
> Does the error remain if you read the temperature sensors multiple times?
> Does rebooting your notebook help?

I have tried everything, but it's not working well

cyear, 2024-09-10:
> Does the error remain if you read the temperature sensors multiple times?
> Does rebooting your notebook help?

Do you know how to control the speed of this fan？

cyear, 2024-09-10:
> Does the error remain if you read the temperature sensors multiple times? Does rebooting your notebook help?

```
[  131.380157] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 57607
[  131.459042] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 0
[  131.533991] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 65
[  131.607987] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28976
[  148.246882] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 56583
[  148.315814] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 21510
[  148.395043] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 57
[  148.469630] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28976
[  179.007850] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 56071
[  179.088030] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 0
[  179.162997] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 95
[  179.243455] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28977
[  183.182289] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 17417
[  183.253842] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 18697
[  183.324705] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 94
[  183.390645] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28978
[  185.320369] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 778
[  185.391728] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 1546
[  185.458425] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 95
[  185.528622] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28978
[  186.492610] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 23562
[  186.570689] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 25098
[  186.638285] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 95
[  186.711552] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28978
[  187.506009] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 45834
[  187.581915] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 50442
[  187.653290] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 95
[  187.724618] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28978
[  188.797314] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 30987
[  188.867883] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 28939
[  188.939951] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 95
[  189.008252] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28978
[  190.547396] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 57611
[  190.617527] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 55307
[  190.688459] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 95
[  190.757940] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28978
[  192.283788] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 31244
[  192.370542] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 31244
[  192.439435] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 95
[  192.507478] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28978
[  194.083166] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 7693
[  194.150831] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 7693
[  194.220849] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 95
[  194.301290] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28978
[  195.328267] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 31245
[  195.402808] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 26893
[  195.467578] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 95
[  195.534603] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28978
[  196.551014] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 55821
[  196.621126] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 48397
[  196.691405] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 95
[  196.760693] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28978
[  198.351824] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 24334
[  198.427371] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 22798
[  198.495810] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 95
[  198.560520] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28978
[  201.968907] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 35343
[  202.033764] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 33295
[  202.103832] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 95
[  202.171847] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28978
[  317.457716] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1124 VALUE 20755
[  317.528388] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1132 VALUE 787
[  317.599369] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1086 VALUE 90
[  317.671786] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: REG 1103 VALUE 28978
[  322.337740] tun: Universal TUN/TAP device driver, 1.6
[... cut ...]
in0:           0.00 V  
power1:           N/A  (max = 100.00 W)
energy1:       9.09 kJ

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +43.0°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +59.9°C  (low  = -273.1°C, high = +89.8°C)
                       (crit = +94.8°C)
Sensor 1:     +59.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +36.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +89.0°C  

nian:uniwill-laptop/ $                       
```

Wer-Wolf, 2024-09-10:
[uniwill-laptop.zip](https://github.com/user-attachments/files/16944633/uniwill-laptop.zip)

The driver uses the standard hwmon pwm interface as described  [here](https://www.kernel.org/doc/Documentation/hwmon/sysfs-interface).

You might have to load the kernel module with `sudo insmod uniwill-laptop.ko force_pwm_enable=1` to enable the pwm interface.

cyear, 2024-09-10:
> [uniwill-笔记本电脑.zip](https://github.com/user-attachments/files/16944633/uniwill-laptop.zip)
> 
> [该驱动程序使用此处](https://www.kernel.org/doc/Documentation/hwmon/sysfs-interface)描述的标准 hwmon pwm 接口 。
> 
> 您可能必须加载内核模块才能`sudo insmod uniwill-laptop.ko force_pwm_enable=1`启用 pwm 接口。

Currently except GPU: +28979.0°C, normal display

cyear, 2024-09-10:
> 只是一个问题：你能以某种方式用英特尔控制台控制风扇速度吗？

Currently I only use the Intel NUC console on Windows to control the fan speed

Wer-Wolf, 2024-09-10:
Ok

cyear, 2024-09-10:
> 风扇控制接口能用吗？

> 风扇控制接口能用吗？

Sorry, I don't know how to control the fan using pwm in linux

Wer-Wolf, 2024-09-10:
> [uniwill-laptop.zip](https://github.com/user-attachments/files/16944633/uniwill-laptop.zip)
>
> The driver uses the standard hwmon pwm interface as described [...]

Ok, please:

1. Compile this module and load it using `sudo insmod uniwill-laptop.ko force_pwm_enable=1`
2. Install `fancontrol`
3. Run `pwmconfig` and share the results

cyear, 2024-09-10:
> > [uniwill-laptop.zip](https://github.com/user-attachments/files/16944633/uniwill-laptop.zip)
> > The driver uses the standard hwmon pwm interface as described [here](https://www.kernel.org/doc/Documentation/hwmon/sysfs-interface).驱动程序使用标准 hwmon pwm 接口，[如此处](https://www.kernel.org/doc/Documentation/hwmon/sysfs-interface)所述。
> > You might have to load the kernel module with `sudo insmod uniwill-laptop.ko force_pwm_enable=1` to enable the pwm interface.您可能必须加载内核 `sudo insmod uniwill-laptop.ko force_pwm_enable=1` 模块才能启用 pwm 接口。
> 
> Ok, please: 好的，请：
> 
> 1. Compile this module and load it using `sudo insmod uniwill-laptop.ko force_pwm_enable=1`编译此模块并使用 `sudo insmod uniwill-laptop.ko force_pwm_enable=1`
> 2. Install `fancontrol`安装 `fancontrol`
> 3. Run `pwmconfig` and share the results运行 `pwmconfig` 并共享结果

```
iceh2o /sys/class/hwmon/hwmon9 # pwmconfig
# pwmconfig version 3.6.0
This program will search your sensors for pulse width modulation (pwm)
controls, and test each one to see if it controls a fan on
your motherboard. Note that many motherboards do not have pwm
circuitry installed, even if your sensor chip supports pwm.

We will attempt to briefly stop each fan using the pwm controls.
The program will attempt to restore each fan to full speed
after testing. However, it is ** very important ** that you
physically verify that the fans have been to full speed
after the program has completed.

Found the following devices:
   hwmon0 is AC0
   hwmon1 is acpitz
   hwmon2 is BAT0
   hwmon3 is nvme
   hwmon4 is coretemp
   hwmon5 is iwlwifi_1
   hwmon6 is ucsi_source_psy_USBC000:001
   hwmon7 is i915
   hwmon8 is ucsi_source_psy_USBC000:002
   hwmon9 is uniwill

Found the following PWM controls:
   hwmon9/pwm1           current value: 56965
   hwmon9/pwm2           current value: 214

Giving the fans some time to reach full speed...
Found the following fan sensors:
   hwmon9/fan1_input     current speed: 5297 RPM
   hwmon9/fan2_input     current speed: 5246 RPM

Warning!!! This program will stop your fans, one at a time,
for approximately 5 seconds each!!!
This may cause your processor temperature to rise!!!
If you do not want to do this hit control-C now!!!
Hit return to continue: 

Testing pwm control hwmon9/pwm1 ...
  hwmon9/fan1_input ... speed was 5297 now 0
    It appears that fan hwmon9/fan1_input
    is controlled by pwm hwmon9/pwm1
Would you like to generate a detailed correlation (y)? y
Note: If you had gnuplot installed, I could generate a graphical plot.
    PWM 255 FAN 5337
    PWM 240 FAN 5109
    PWM 225 FAN 4867
    PWM 210 FAN 4657
    PWM 195 FAN 4409
    PWM 180 FAN 4186
    PWM 165 FAN 3927
    PWM 150 FAN 3648
    PWM 135 FAN 3358
    PWM 120 FAN 3071
    PWM 105 FAN 2753
    PWM 90 FAN 2398
    PWM 75 FAN 2028
    PWM 60 FAN 1678
    PWM 45 FAN 1267
    PWM 30 FAN 844
    PWM 28 FAN 758
    PWM 26 FAN 715
    PWM 24 FAN 635
    PWM 22 FAN 584
    PWM 20 FAN 0
    Fan Stopped at PWM = 20

  hwmon9/fan2_input ... speed was 5246 now 5233
    no correlation

Testing pwm control hwmon9/pwm2 ...
  hwmon9/fan1_input ... speed was 5297 now 5337
    no correlation
  hwmon9/fan2_input ... speed was 5246 now 0
    It appears that fan hwmon9/fan2_input
    is controlled by pwm hwmon9/pwm2
    Fan hwmon9/fan2_input has not returned to speed, please investigate!

Testing is complete.
Please verify that all fans have returned to their normal speed.

The fancontrol script can automatically respond to temperature changes
of your system by changing fanspeeds.
Do you want to set up its configuration file now (y)? y
What should be the path to your fancontrol config file (/etc/fancontrol)? 

Select fan output to configure, or other action:
1) hwmon9/pwm2	      3) Change INTERVAL     5) Save and quit
2) hwmon9/pwm1	      4) Just quit	     6) Show configuration
select (1-n): 6

Common Settings:
INTERVAL=10

Settings of hwmon9/pwm2:
  Depends on 
  Controls 
  MINTEMP=
  MAXTEMP=
  MINSTART=
  MINSTOP=

Settings of hwmon9/pwm1:
  Depends on 
  Controls 
  MINTEMP=
  MAXTEMP=
  MINSTART=
  MINSTOP=

Select fan output to configure, or other action:
1) hwmon9/pwm2	      3) Change INTERVAL     5) Save and quit
2) hwmon9/pwm1	      4) Just quit	     6) Show configuration
select (1-n): 
```
OMG, this is the first time I feel like the fan is controllable, thank you so much

cyear, 2024-09-10:
> The driver is still buggy, bit the first results are indeed promising.驱动程序仍然有问题，但初步结果确实很有希望。

> The driver is still buggy, bit the first results are indeed promising.驱动程序仍然有问题，但初步结果确实很有希望。

YES！

Wer-Wolf, 2024-09-10:
I will try to stabilize the second temperature sensor and the fan control interface. This might take a while.

cyear, 2024-09-10:
> [uniwill-laptop.zip](https://github.com/user-attachments/files/16945082/uniwill-laptop.zip)
> 
> Can you test this module too? It should print a message similar to "Fan control reg: X" during loading. Can you show me this message.您也可以测试此模块吗？在加载过程中，它应该打印类似于 “Fan control reg： X” 的消息。您能给我看一下这条信息吗？

```
[ 2833.611073] Setting dangerous option force_pwm_enable - tainting kernel
[ 2833.697897] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: Project ID: 8
[ 2833.766639] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: Fan control reg: 66
[ 2891.368399] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: Project ID: 8
[ 2891.434567] uniwill ABBC0F6F-8EA1-11D1-00A0-C90629100000: Fan control reg: 66
```

Wer-Wolf, 2024-09-10:
I see, the EC seems to support a different fan control interface. I will integrate it into the driver and send you another prototype soon which should work without having to set `force_pwm_enable`.

cyear, 2024-09-11:
> I see, the EC seems to support a different fan control interface. I will integrate it into the driver and send you another prototype soon which should work without having to set `force_pwm_enable`.

If I want to control the fan speed on Windows, can I control it through `ec_probe`?

Wer-Wolf, 2024-09-11:
I do not think so, but you can use the WMI interface to do it. If you want, i can write some documentation on how to use the WMI interface.

Wer-Wolf, 2024-09-11:
I created a [ticket for the other driver](https://gitlab.com/tuxedocomputers/development/packages/tuxedo-drivers/-/issues/138) to better understand the EC interface.

The reason for this is that both drivers use different methods of controlling the fan, and i would like to know which one is correct.

cyear, 2024-09-12:
> I created a [ticket for the other driver](https://gitlab.com/tuxedocomputers/development/packages/tuxedo-drivers/-/issues/138) to better understand the EC interface.
>
> The reason for this is that [...]

I have created a session with Intel technical support, although there is little hope, I will let you know if there is any good news

cyear, 2024-09-12:
> I created a [ticket for the other driver](https://gitlab.com/tuxedocomputers/development/packages/tuxedo-drivers/-/issues/138) to better understand the EC interface.
>
> The reason for this is that [...]

If only I could see what actions the NUC console has taken on the device

Wer-Wolf, 2024-09-12:
I will write some documentation tomorrow on how to use the WMI interface for this.

This should allow you to read and write EC RAM.

cyear, 2024-09-12:
> I will write some documentation tomorrow on how to use the WMI interface for this.
> 
> This should allow you to read and write EC RAM.

I don't quite understand `WMI`, is it `Windows Management Instrumentation`? I don't know much about programming, but if possible, I would be willing to give it a try

Wer-Wolf, 2024-09-12:
Yes, you can use it with powershell under Windows.

cyear, 2024-09-12:
> Yes, you can use it with powershell under Windows.

Unfortunately, Intel cannot provide any assistance regarding fans

cyear, 2024-09-12:
> Thats unfortunate, then we have to do some reverse engineering ourself.
> 
> I will provide you with a set of powershell scrips which you can use to interact with the EC under Windows.
> 
> I will also provide you with a set of register values which need to be tested. This way you dont have to recompile a kernel module each time (and i can meanwhile stabilize the driver xd).

Okay, Thanks for your help

Wer-Wolf, 2024-09-12:
I created a separate repository [here](https://github.com/Wer-Wolf/uniwill-laptop). The wiki tab contains a short guide on how to access the WMI interface.

Try to use this interface to read the following registers:
- `0x043E`
- `0x044F`
- `0x0464`
- `0x0465`
- `0x046C`
- `0x046D`
- `0x0742`
- `0x0751`
- `0x075B`
- `0x075C`
- `0x0765`
- `0x0766`
- `0x078E`
- `0x1804`
- `0x1809`

cyear, 2024-09-12:
> 我[在这里](https://github.com/Wer-Wolf/uniwill-laptop)创建了一个单独的存储库。Wiki 选项卡包含有关如何访问 WMI 接口的简短指南。
> 
> 尝试使用该接口读取以下寄存器：
> 
> * `0x043E`
> * `0x044F`
> * `0x0464`
> * `0x0465`
> * `0x046C`
> * `0x046D`
> * `0x0742`
> * `0x0751`
> * `0x075B`
> * `0x075C`
> * `0x0765`
> * `0x0766`
> * `0x078E`
> * `0x1804`
> * `0x1809`

```
PS C:\WINDOWS\system32>  $obj = Get-CimInstance -Namespace root/wmi -ClassName AcpiTest_MULong
PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x000001000000043E}
Invoke-CimMethod : 无法将“System.Object[]”转换为参数“InputObject”所需的类型“Microsoft.Management.Infrastructure.Ci
mInstance”。不支持所指定的方法。
所在位置 行:1 字符: 31
+ Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments ...
+                               ~~~~
    + CategoryInfo          : InvalidArgument: (:) [Invoke-CimMethod]，ParameterBindingException
    + FullyQualifiedErrorId : CannotConvertArgument,Microsoft.Management.Infrastructure.CimCmdlets.InvokeCimMethodComm
   and

PS C:\WINDOWS\system32>
```
I don't quite understand what happened

cyear, 2024-09-12:
> 我创建[了](https://github.com/Wer-Wolf/uniwill-laptop)一个单独的存储库。Wiki 选项卡包含有关如何访问 WMI 接口的简单指南。
> 
> 尝试使用该接口读取以下记录：
> 
> * `0x043E`
> * `0x044F`
> * `0x0464`
> * `0x0465`
> * `0x046C`
> * `0x046D`
> * `0x0742`
> * `0x0751`
> * `0x075B`
> * `0x075C`
> * `0x0765`
> * `0x0766`
> * `0x078E`
> * `0x1804`
> * `0x1809`

```
PS C:\WINDOWS\system32> $obj.GetType()

IsPublic IsSerial Name                                     BaseType
-------- -------- ----                                     --------
True     True     Object[]                                 System.Array

PS C:\WINDOWS\system32> $obj

Active InstanceName     PSComputerName
------ ------------     --------------
  True ACPI\PNP0C14\1_0
  True ACPI\PNP0C14\1_1
  True ACPI\PNP0C14\1_2
  True ACPI\PNP0C14\1_3
  True ACPI\PNP0C14\1_4
  True ACPI\PNP0C14\1_5
  True ACPI\PNP0C14\1_6
  True ACPI\PNP0C14\1_7
  True ACPI\PNP0C14\1_8
  True ACPI\PNP0C14\1_9

```
what?

cyear, 2024-09-12:
> 我创建[了](https://github.com/Wer-Wolf/uniwill-laptop)一个单独的存储库。Wiki 选项卡包含有关如何访问 WMI 接口的简单指南。
> 
> 尝试使用该接口读取以下记录：
> 
> * `0x043E`
> * `0x044F`
> * `0x0464`
> * `0x0465`
> * `0x046C`
> * `0x046D`
> * `0x0742`
> * `0x0751`
> * `0x075B`
> * `0x075C`
> * `0x0765`
> * `0x0766`
> * `0x078E`
> * `0x1804`
> * `0x1809`

```
PS C:\WINDOWS\system32> $obj = $obj | Select-Object -First 1
PS C:\WINDOWS\system32> $obj

Active InstanceName     PSComputerName
------ ------------     --------------
  True ACPI\PNP0C14\1_0

PS C:\WINDOWS\system32> $obj.GetType()

IsPublic IsSerial Name                                     BaseType
-------- -------- ----                                     --------
True     True     CimInstance                              System.Object

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x000001000000043E}

Return ReturnValue PSComputerName
------ ----------- --------------
    52        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x000001000000044F}

Return ReturnValue PSComputerName
------ ----------- --------------
 28952        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x0000010000000464}

Return ReturnValue PSComputerName
------ ----------- --------------
 64775        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x0000010000000465}

Return ReturnValue PSComputerName
------ ----------- --------------
   253        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x000001000000046C}

Return ReturnValue PSComputerName
------ ----------- --------------
     0        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x000001000000046D}

Return ReturnValue PSComputerName
------ ----------- --------------
     0        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x0000010000000742}

Return ReturnValue PSComputerName
------ ----------- --------------
     2        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x0000010000000751}

Return ReturnValue PSComputerName
------ ----------- --------------
 27664        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x000001000000075B}

Return ReturnValue PSComputerName
------ ----------- --------------
    60        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x000001000000075C}

Return ReturnValue PSComputerName
------ ----------- --------------
     0        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x0000010000000765}

Return ReturnValue PSComputerName
------ ----------- --------------
 33792        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x0000010000000766}

Return ReturnValue PSComputerName
------ ----------- --------------
   132        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x000001000000078E}

Return ReturnValue PSComputerName
------ ----------- --------------
    66        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x0000010000001804}

Return ReturnValue PSComputerName
------ ----------- --------------
 42812        True

PS C:\WINDOWS\system32> Invoke-CimMethod -InputObject $obj -MethodName GetSetULong -Arguments @{Data = 0x0000010000001809}

Return ReturnValue PSComputerName
------ ----------- --------------
     0        True

PS C:\WINDOWS\system32>
```
I don't know if my operation is correct

Wer-Wolf, 2024-09-12:
I think it is correct.

Wer-Wolf, 2024-09-12:
Could it be that your notebook has two fans and one was not spinning during the last WMI operation?

[149 more replies]

closed 2024-09-20 by cyear