lm-sensors/lm-sensors #510: Lenovo Thinkpad L540 Linux Mint 20.2 fan1 always 0

insomniacno1, 2024-09-05:
Hi, I have read numerous writeups and suggestions to how to get lm-sensors to recognize the fan in my Lenovo Thinkpad L540 model 20aua0f4jp serial R902K1R2  but always end up with fan1 is 0 rpm.

Can you help me figure out how to get this working?

Fan is working, I took the back panel off and can confirm that the is running. It also have more than 3 cables so should have sensor abillity.

Kernel 5.15.0-119 - generic

insomniacno1, 2024-09-05:
[image]

insomniacno1, 2024-09-05:
Ok so continue here, hope someone can share light on how to fix this, fan runs but doesn't show anything but 0 rpm.

There is no setting in bios to change anything fan related, bios 1.93(Newest).

Thank you in advance

Jan

insomniacno1, 2024-09-22:
sudo lsmod output

[image]
[image]
[image]
[image]

sudo sensors-detect output
[image]
[image]
[image]
[image]

insomniacno1, 2024-09-22:
sudo lshw output

th-thinkpad-x300            
    description: Notebook
    product: 20AUA0F4JP (LENOVO_MT_20AU_BU_Think_FM_ThinkPad L540)
    vendor: LENOVO
    version: ThinkPad L540
    serial: R902K1R2
    width: 64 bits
    capabilities: smbios-2.7 dmi-2.7 smp vsyscall32
    configuration: administrator_password=disabled chassis=notebook family=ThinkPad L540 power-on_password=disabled sku=LENOVO_MT_20AU_BU_Think_FM_ThinkPad L540 uuid=8167E192-F252-CB11-8EA7-BF5E55C761D2
  *-core
       description: Motherboard
       product: 20AUA0F4JP
       vendor: LENOVO
       physical id: 0
       version: 0B98401 Pro
       serial: VQ2MM46H2BU
       slot: Not Available
     *-cpu
          description: CPU
          product: Intel(R) Core(TM) i5-4200M CPU @ 2.50GHz
          vendor: Intel Corp.
          physical id: 0
          bus info: cpu@0
          version: Intel(R) Core(TM) i5-4200M CPU @ 2.50GHz
          serial: To Be Filled By O.E.M.
          slot: CPU Socket - U3E1
          size: 2500MHz
          capacity: 3100MHz
          width: 64 bits
          clock: 100MHz
          capabilities: lm fpu fpu_exception wp vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp x86-64 constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm ida arat pln pts md_clear flush_l1d cpufreq
          configuration: cores=2 enabledcores=2 threads=4
        *-cache:0
             description: L1 cache
             physical id: 2
             slot: L1-Cache
             size: 32KiB
             capacity: 32KiB
             capabilities: asynchronous internal write-back instruction
             configuration: level=1
        *-cache:1
             description: L2 cache
             physical id: 3
             slot: L2-Cache
             size: 256KiB
             capacity: 256KiB
             capabilities: asynchronous internal write-back unified
             configuration: level=2
        *-cache:2
             description: L3 cache
             physical id: 4
             slot: L3-Cache
             size: 3MiB
             capacity: 3MiB
             capabilities: asynchronous internal write-back unified
             configuration: level=3
     *-cache
          description: L1 cache
          physical id: 1
          slot: L1-Cache
          size: 32KiB
          capacity: 32KiB
          capabilities: asynchronous internal write-back data
          configuration: level=1
     *-memory
          description: System Memory
          physical id: 5
          slot: System board or motherboard
[... 372 lines cut ...]
                physical id: 3
                bus info: scsi@4:0.0.0,3
                logical name: /dev/sda3
                size: 173GiB
                capacity: 173GiB
                capabilities: primary extended partitioned partitioned:extended
              *-logicalvolume
                   description: EXT4 volume
                   vendor: Linux
                   physical id: 5
                   logical name: /dev/sda5
                   logical name: /
                   version: 1.0
                   serial: 615456fc-8e14-4b2b-a856-400293ac00a3
                   size: 173GiB
                   capacity: 173GiB
                   capabilities: journaled extended_attributes large_files huge_files dir_nlink recover 64bit extents ext4 ext2 initialized
                   configuration: created=2021-11-28 17:34:02 filesystem=ext4 lastmountpoint=/ modified=2024-09-22 09:11:26 mount.fstype=ext4 mount.options=rw,relatime,errors=remount-ro mounted=2024-09-22 09:11:27 state=mounted
  *-battery
       product: 45N1769
       vendor: SANYO
       physical id: 1
       slot: Front
       capacity: 56160mWh
       configuration: voltage=10.8V

insomniacno1, 2024-09-22:
sudo lspci output

00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor DRAM Controller (rev 06)
00:02.0 VGA compatible controller: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)
00:03.0 Audio device: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor HD Audio Controller (rev 06)
00:14.0 USB controller: Intel Corporation 8 Series/C220 Series Chipset Family USB xHCI (rev 05)
00:16.0 Communication controller: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 (rev 04)
00:19.0 Ethernet controller: Intel Corporation Ethernet Connection I217-V (rev 05)
00:1b.0 Audio device: Intel Corporation 8 Series/C220 Series Chipset High Definition Audio Controller (rev 05)
00:1c.0 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 (rev d5)
00:1c.1 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #2 (rev d5)
00:1c.2 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #3 (rev d5)
00:1c.4 PCI bridge: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #5 (rev d5)
00:1f.0 ISA bridge: Intel Corporation HM86 Express LPC Controller (rev 05)
00:1f.2 SATA controller: Intel Corporation 8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode] (rev 05)
00:1f.3 SMBus: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller (rev 05)
02:00.0 Network controller: Intel Corporation Wireless 7260 (rev 83)
08:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5227 PCI Express Card Reader (rev 01)

insomniacno1, 2024-09-22:
sensors-detect finds Intel digital thermal sensor, could that be the problem, do I need to add something to grub on boot?

insomniacno1, 2024-09-24:
A bit more investigation seem to find that my motherboard has a nuvoTon chip. but it is not found by lm-sensors, I cant tell which model number it is as all photos are blurred on this part of the system board

insomniacno1, 2024-09-26:
Bios photo
[image]

insomniacno1, 2024-10-08:
I have found out that troubadix tpfc 0.62L can detect my fan and manipulate it in Windows 10, this means that the fan can be detected and it also mean that this is a lm-sensor problem not the fan!

Could you please help me fix this, newer version of linux with kernel 6.1 is very unstable, I know you fixed this issue for someone else with Novuton controller, but send the fix upstream.

Thank you in advance

Jan

insomniacno1, 2026-02-22:
February 22 2026 - Problem persist

salmundani, 2026-09-15:
This isn't an lm-sensors bug. lm-sensors just prints what the kernel's thinkpad_acpi driver gives it, and on the L440/L540 the driver reads the wrong EC registers. These machines keep the fan status at EC offset 0x93 and the speed at 0x95, not at 0x2F/0x84 like most ThinkPads. The kernel has handled that layout since 6.8 for a few other models (L13 Yoga, L380, L390, Edge E330), the L440/L540 just weren't on the list.

I sent a patch to add them: https://lore.kernel.org/all/20260915021154.23252-1-salmundani@gmail.com/T/#u

On my L440 (BIOS 1.93) sensors now shows ~1900 RPM idle and ~2900 RPM under load.

The L540 has the same BIOS and EC firmware so it should be covered too, but I haven't got one to try. If you want to check yours, with the fan running:

```
sudo modprobe ec_sys
sudo dd if=/sys/kernel/debug/ec/ec0/io bs=1 skip=$((0x93)) count=1 2>/dev/null | od -An -tu1
sudo dd if=/sys/kernel/debug/ec/ec0/io bs=1 skip=$((0x95)) count=1 2>/dev/null | od -An -tu1
```

Mine prints 4 for the first and 255 for the second at idle, going down to ~166 under load. Same numbers on yours means the patch will work there.

If you can build a kernel with the patch and it works on your L540, a Tested-by reply on the kernel mailing thread would be great, since I can only vouch for the L440.