lm-sensors/lm-sensors #520: "sensors" should be smart enough not to emit chip or adapter name if zero sensors

mikkorantalainen, 2024-11-19:
The command line program `sensors` emits chip and adapter names even if no sensors can be found via the chip. For example:

```
$ sensors
iwlwifi_1-virtual-0
Adapter: Virtual device

acpitz-acpi-0
Adapter: ACPI interface

nct6687-isa-0a20
...
```

This may happen because all the sensors have been ignored in the config file or because no sensors were detected. As an alternative, the config file should support ignoring the whole chip to remove it from the output but it would make more sense to automatically leave the chip from the output if no usable sensors exist.

If you consider disabling the chip via the config file better as better option, maybe implement it with the following syntax?

```
chip "iwlwifi_1-virtual-0"
ignore *
```

That would also extend to future improvement where one could ignore all temperature sensors (`ignore temp*`) without ignoring the whole chip.

https://github.com/lm-sensors/lm-sensors/blob/42f240d2a457834bcbdf4dc8b57237f97b5f5854/prog/sensors/main.c#L163

etieng, 2026-08-16:
Today I come to this issue because `sensors -j`  produce an incorrectly formatted json with all sensors ignored in a chip : 
*etc/sensors.d/myconf.conf* :
```
chip "acpitz-acpi-*"
        ignore  temp1
        ignore  temp2
```
*sensors -j* : 
`   "acpitz-acpi-0":{
      "Adapter": "ACPI interface",
   },
`

 Which indeed produce parsing errors in scripts like this https://codeberg.org/LINET-Services/linet-lmsensors ;-)

etieng, 2026-08-16:
Sounds duplicated of #513 and #372