lm-sensors/lm-sensors #521: sensors-detect reports "unknown chip with ID 0xd592" instead of loading module nct6683

mikkorantalainen, 2024-11-21:
Running `sensors-detect` on Ubuntu 22.04 LTS results in output

    Found unknown chip with ID 0xd592

but no module is loaded. If one knows that this chip is supported by module `nct6683` and loads this module before running `sensors` the output works as expected.

The `sensors-detect` should be smart enough to load this module automatically.

I think the actual chip support is enabled via this line in Linux kernel:
https://github.com/torvalds/linux/blob/43fb83c17ba2d63dfb798f0be7453ed55ca3f9c2/drivers/hwmon/nct6683.c#L46

I found this issue with MSI motherboard called "PRO Z790-P WIFI DDR4".

zmascetta, 2024-11-28:
There are reports in other issues with this family of chips (6686, 6687, etc.) never being detected unless you run `modprobe 6683`. I have a 6686D and run into that issue as well. If you just want to automatically load this driver on boot, you can add it to the modules by using `sudo sh -c 'echo "nct6683" >> /etc/modules'`.

In my case, I wanted to load the sensor in order to control fan speed. If that is what you want to do as well, the 6683 driver in the kernel currently doesn't support controlling PWM. To get around this, I installed [this driver](https://github.com/Fred78290/nct6687d), which adds PWM support. I am not sure the exact chip in your motherboard, but you could try it out and see if it works.

iznobe, 2025-04-05:
Hello , o got same issue .

i must add manually " nct6683 " to /etc/modules to have lmsensors working .

Now works as expected .
i was looking for a very long time to have this working .

akaguny, 2025-05-06:
@iznobe

> ERROR: Can't get value of subfeature temp7_min: I/O error

if you allready install https://github.com/Fred78290/nct6687d

try to remove `nct6683` from `/etc/modules`

also /etc/modules is deprecated (in your file should exist desclimer) 

as variant is blacklist 6683 https://github.com/Fred78290/nct6687d/issues/96#issuecomment-2573654408