lm-sensors/lm-sensors #525: no sensors detected on GIGABYTE X670 AORUS ELITE AX

alfonz19, 2025-01-31:
I'm exceptional newbie in this area, and while I read and tried a lot, I'm completely lost. Can someone recommend, what I can do to get some readings from this motherboard and cpu.

sensors detect using all default choices:

```
Sorry, no sensors were detected.
Either your system has no sensors, or they are not supported, or
they are connected to an I2C or SMBus adapter that is not
supported. If you find out what chips are on your board, check
https://hwmon.wiki.kernel.org/device_support_status for driver status.
```

inxi -F (shortened
```
System:
  Host: VbDesktop2 Kernel: 6.11.0-13-generic arch: x86_64 bits: 64
  Desktop: KDE Plasma v: 6.1.5 Distro: Kubuntu 24.10 (Oracular Oriole)
Machine:
  Type: Desktop Mobo: Gigabyte model: X670 AORUS ELITE AX
    serial: <superuser required> UEFI: American Megatrends LLC. v: F32h
    date: 12/19/2024
CPU:
  Info: 8-core model: AMD Ryzen 7 9800X3D bits: 64 type: MCP cache: L2: 8 MiB
  Speed (MHz): avg: 3083 min/max: 600/5269 cores: 1: 2983 2: 2983 3: 2983
    4: 2983 5: 3679 6: 3619 7: 2983 8: 2452
Graphics:
  Device-1: AMD Navi 31 [Radeon RX 7900 XT/7900 XTX/7900 GRE/7900M]
```

and I won't get more readings than:

```
mt7921_phy0-pci-1100
Adapter: PCI adapter
temp1:        +40.0°C  

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +46.2°C  

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8°C  

amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:      192.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 3600 RPM)
edge:         +58.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +66.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +74.0°C  (crit = +108.0°C, hyst = -273.1°C)
                       (emerg = +113.0°C)
PPT:          39.00 W  (cap = 303.00 W)

gigabyte_wmi-virtual-0
Adapter: Virtual device
temp1:        +31.0°C  
temp2:        +63.0°C  
temp3:        +46.0°C  
temp4:        +43.0°C  
temp5:        +40.0°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +44.9°C  (low  = -20.1°C, high = +83.8°C)
                       (crit = +88.8°C)
Sensor 2:     +73.8°C  

```

To be honest I have no idea if chip used for this card is Nuvoton NCT6799D-R or it87, I saw sources claiming both.

I found 2 sources about this very card (manufacturer not mentioned or different):

https://forum.manjaro.org/t/sensors-for-fan-control-on-x670-aorus-elite-ax/137600
https://github.com/lm-sensors/lm-sensors/issues/416

but neither did help me.

Some more info:

```
lsmod | grep "nct6775\|coretemp"
nct6775_core          106496  0

lsmod | grep "nct6775\|k10temp"
nct6775_core          106496  0
k10temp                16384  0

sudo dmesg | grep nct6775

sudo dmesg | grep nct6775_core

lsmod | grep nct6775
nct6775_core          106496  0
```

surprisingly (for me) there are 2 nct6775 modules: nct6775 and nct6775_core:

```
modinfo nct6775
filename:       /lib/modules/6.11.0-13-generic/kernel/drivers/hwmon/nct6775.ko.zst
import_ns:      HWMON_NCT6775
license:        GPL
description:    Platform driver for NCT6775F and compatible chips
author:         Guenter Roeck <linux@roeck-us.net>
srcversion:     947EC0B0E2B694697E0EEA3
depends:        hwmon-vid,nct6775-core
retpoline:      Y
intree:         Y
name:           nct6775
vermagic:       6.11.0-13-generic SMP preempt mod_unload modversions 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        3F:6D:EE:43:8C:BC:38:16:19:4A:7A:A9:AF:B4:EB:6A:E8:1B:85:C9
sig_hashalgo:   sha512
signature:      60:19:1D:F0:23:51:E5:E4:18:6D:FD:3B:46:D0:C9:01:F9:62:64:74:
		25:99:32:51:C3:0E:01:1D:1F:C7:A1:A9:7A:B5:63:32:6D:46:A0:B7:
		4E:40:D3:EF:09:2F:4B:62:9C:E1:CD:12:F3:DB:AD:59:16:D7:5D:80:
		40:CA:6C:FD:1D:B6:66:97:8B:1C:EB:83:38:EB:10:66:1C:00:49:D8:
		48:38:9C:C2:93:2E:AD:A6:50:00:BF:50:16:DF:F0:F2:A6:75:77:BF:
		22:58:6F:50:5A:F8:25:FF:E1:00:DC:C2:17:3E:DE:5B:6F:25:00:EB:
		26:27:ED:B7:BC:32:55:1D:D7:A3:E1:EF:18:0C:DB:A1:20:9E:19:F5:
		ED:94:77:0F:63:4B:39:3C:44:F8:CC:9C:36:1F:E6:3B:2B:FB:D9:E2:
		F9:AB:F4:E6:C4:B3:DF:59:EE:35:8F:7F:B3:7F:B2:81:C1:20:09:FD:
		FC:83:A3:CE:CE:F6:FA:C3:AE:6B:66:37:D0:FB:16:0A:F8:D1:ED:92:
		34:47:4A:50:94:2F:21:10:96:41:19:60:6A:F1:9E:D0:8C:E8:C5:42:
		21:32:03:0B:40:B4:F8:9F:37:BA:6B:48:91:94:5C:96:96:A2:5C:B9:
		6E:09:9C:D9:F0:F2:64:1C:50:C6:98:FF:DF:9C:8D:0F:E6:59:8B:CE:
		FF:1C:24:FC:FF:39:2C:CB:A0:7C:FA:DF:EB:19:87:D2:7D:BF:6A:01:
		69:E8:95:74:1F:01:48:1B:82:C2:9F:5A:99:8E:E7:04:A2:CC:A0:FD:
		84:D6:ED:3C:93:1E:2D:46:F4:56:FE:C7:F4:AC:0C:D4:35:FE:E3:AF:
		4B:90:5E:30:D4:80:91:F4:EF:2B:F9:D1:77:04:0B:56:5B:8E:6F:F4:
		8F:0A:C0:CA:6E:DE:EB:82:B0:A4:9B:1D:18:CF:17:2B:47:00:3B:4A:
		49:F6:68:F7:DB:24:ED:CC:BA:C1:9E:57:C5:D7:17:4F:10:FB:05:D5:
		9D:BD:57:E3:23:12:2C:67:C6:83:77:CD:1F:FD:03:1A:8E:25:0C:39:
		0A:C1:FF:D3:98:0F:52:78:DF:1E:DC:6E:20:54:1D:D8:C0:78:CC:36:
		88:44:8A:8E:32:E2:A2:D6:BD:F1:CE:40:F9:7C:8B:9E:13:F4:61:AF:
		F7:9C:EF:7C:16:99:08:00:B4:FE:A9:75:B8:2B:C9:52:B4:64:5D:22:
		7A:ED:2E:C1:53:67:2C:C0:24:74:8A:78:01:24:1A:A4:8A:F0:42:3F:
		A2:FD:F5:79:12:A9:55:A7:92:66:98:FD:02:14:53:40:A4:72:DA:05:
		74:1D:9A:29:F0:0D:EF:9B:82:F0:3C:A0
parm:           force_id:Override the detected device ID (ushort)
parm:           fan_debounce:Enable debouncing for fan RPM signal (ushort)

modinfo nct6775_core

filename:       /lib/modules/6.11.0-13-generic/kernel/drivers/hwmon/nct6775-core.ko.zst
license:        GPL
description:    Core driver for NCT6775F and compatible chips
author:         Guenter Roeck <linux@roeck-us.net>
srcversion:     E533A571A44D489E3A69FFE
depends:        
retpoline:      Y
intree:         Y
name:           nct6775_core
vermagic:       6.11.0-13-generic SMP preempt mod_unload modversions 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        3F:6D:EE:43:8C:BC:38:16:19:4A:7A:A9:AF:B4:EB:6A:E8:1B:85:C9
sig_hashalgo:   sha512
signature:      4B:3F:CB:60:2C:43:83:FA:EF:B4:0B:D7:35:43:BF:DE:E0:B6:1E:23:
		9C:5E:CC:17:5D:2F:83:DB:D0:6B:69:CD:41:DF:C3:31:EB:2E:AD:87:
		72:25:BB:D3:23:1F:45:02:D5:89:71:96:BF:A0:90:1D:D7:EC:BC:89:
		B5:59:ED:42:B1:D0:21:C6:87:FC:B4:00:84:F8:5C:77:C5:0D:76:05:
		EF:C7:51:F1:90:7E:36:39:BE:6F:61:D8:2A:34:E4:1A:D7:4B:B2:23:
		17:0C:0F:CA:68:FF:E8:D2:9C:73:AF:A8:5A:DA:14:D8:63:72:06:B7:
		4E:8C:A2:DD:B9:84:05:66:4A:A7:AB:16:52:58:D4:95:C1:9B:87:36:
		27:8C:C2:3A:A5:1F:D0:55:BC:D9:B9:1F:E2:95:35:45:29:16:45:EF:
		E8:B3:51:8C:A2:BC:D6:C2:AB:11:67:0E:D0:90:5C:7C:AC:13:EF:63:
		0F:26:CC:84:01:D3:5E:3B:DA:14:06:9C:4D:7A:12:BC:6E:D7:7C:10:
		24:E0:A7:AF:7F:C4:89:7B:E3:07:D6:37:D8:D4:10:1C:13:26:D1:D4:
		30:55:E0:A3:4C:4E:51:8C:66:83:49:37:9B:F1:01:52:6C:B5:32:A6:
		E8:95:72:61:18:FC:81:A8:38:21:D6:DC:E1:90:32:20:A4:0D:A9:7C:
		A8:D6:6A:A0:86:30:37:54:BF:7A:00:2E:E7:A7:7C:39:5A:03:32:6F:
		91:1D:A4:E5:39:F4:4B:63:46:ED:A6:1C:79:55:DF:76:0A:4B:83:7C:
		89:37:0B:9D:8C:6E:1F:22:79:15:8D:66:07:FD:96:55:6B:D5:0D:5F:
		5E:13:22:6A:F6:FF:76:D1:7A:92:B5:85:8C:5E:21:69:28:88:ED:76:
		07:25:4B:E9:70:90:53:8F:5C:23:2E:3E:CA:B2:75:A4:56:B9:AD:48:
		5D:43:C6:9F:31:CB:EB:27:75:18:49:B7:44:A2:FD:69:24:21:9B:A4:
		EE:7C:E7:9B:94:7E:0A:54:3B:69:08:98:55:DC:0D:02:07:EC:F0:07:
		F3:F6:DB:2E:0D:C8:54:7E:F5:7A:6A:42:83:69:C1:65:14:12:61:D3:
		FD:8D:34:19:F8:68:BF:E4:F3:05:0A:62:2B:88:1A:A9:B8:DA:5E:28:
		85:27:4E:82:D5:57:69:D2:F3:41:0A:CC:59:97:27:92:40:C0:21:B6:
		CB:4A:AC:8C:DA:45:96:89:10:C8:8C:CE:1F:3B:1C:87:B9:C6:1E:6D:
		1C:FE:A8:1B:54:3C:47:C0:53:0C:ED:4C:91:EA:01:99:7C:3F:C3:11:
		92:C1:7C:5F:39:61:C2:0F:F4:BB:D7:CB

```

I tried adding kernel parameter `acpi_enforce_resources=lax` and loading kernel module with different parameters, but nothing helped.

Can anyone advice or give some pointer?

thedanbob, 2025-05-02:
I have this motherboard, too. I spotted the IT8689E chip on it which unfortunately doesn't seem to be currently supported by it87: https://docs.kernel.org/hwmon/it87.html

sunbearc22, 2025-07-26:
Try [CoolerControl](https://docs.coolercontrol.org/)?  It worked for me. This project was very useful in the past but over recent years seem unmaintained and dated, except by a few of its forks.  See #526 on project status.