lm-sensors/lm-sensors #526: Project Status

B3rn475, 2025-02-02:
Hi,
there is not simple way to ask this but... __Is the project still alive?__
__IMPORTANT:__ _I know that every time this question is asked it can sound aggressive or presumptuous._

The current state is that:
- No changes/new versions landed in the last 4 years.
- The website is down [see](https://github.com/lm-sensors/lm-sensors/issues/217) and the replacement never actually landed and is marked as deprecated too.
- Many issues have been opened but there was no interaction from the developers.

Is this because with the migration to `sysfs` the complexity has been moved to a different layer and there is no need to evolve the project? Or was the project abandoned?

Or, in other words, should people consider using `lm-sensors` today? Or it is better to search for an alternative?

MarcusPro7, 2025-03-08:
Hi, I was asking myself the same question after installing it on a ClearLinux distribution. 
When I run the `pwmconfig `command, I get the following message: `/usr/bin/pwmconfig: There are no pwm-capable sensor modules installed`. 

I am using an HP Slice Elite G1 with 6.13.4-1548.native

CarliJoy, 2025-04-06:
I also feel the project is dead.

I seems that https://gitlab.com/coolercontrol/coolercontrol looks promising as replacement.

B3rn475, 2025-04-07:
@CarliJoy it looks like they use lm-sensors, at least partially.
https://gitlab.com/coolercontrol/coolercontrol#-hardware-support talks about installing and configuring `lm-sensors`.

CarliJoy, 2025-04-07:
Maeh, you are right. Seems like we stuck with it.
Or did anybody find an alternative or considerable fork?
I saw that for debian a number of patches have been applied.

CarliJoy, 2025-04-08:
I used ChatGPG Deep Research to ask about the project status [[Source](https://chatgpt.com/share/67f58dc2-c748-800c-a707-0c3b5efd18ce)].

The answer shows how many years of engagement was already put into the project. Thank you everybody involved for that 🤗 😍.

The answer mentioned the most notable [fork](https://github.com/hramrach/lm-sensors).

I saw Guenter Roeck [commented](https://github.com/lm-sensors/lm-sensors/issues/3#issuecomment-233141801) in 2016:

> lm-sensors is and will remain down. the hwmon wiki is far from complete, and will be for a while as I have little time to do the work.

Maybe @groeck can tell us more what this means?
My question: If the project is no longer maintained, would it be okay to state that clearly in the projects README?

Also my question for the creator of the fork Michal Suchánek (@hramrach), if he would be okay with trying to merge his changes into the official repo / become kind of maintainer.

Here the Chat GPG Answer in short:

## Question

Can you find out who are the maintainers of lm-sensors?  
The project https://github.com/lm-sensors/lm-sensors seems dead and it would be nice to know who is or was responsible for the project.  

Also consider information of Linux distributions like Debian.  

Also, if you see an alternative or fork `lm-sensors` let me know.  

---

## Answer *(generated by ChatGPT)*

### Maintainers and Status of the LM-Sensors Project

---

### 🧑‍💻 Historical and Current Maintainers

- **Early developers (1998–2003):** The lm-sensors project was originally developed by volunteers in the late 1990s. Key early contributors included **Frodo Looijaard**, **Merlin Hughes**, and **Adrian Baugh**, who wrote initial code for sensor drivers and tools. The project was later maintained by **Philip Edelbrock** (founder of the lm-sensors project and original mailing list host) and **Mark D. Studebaker**, who led development in the early 2000s.

- **Mid-2000s maintainers:** Around 2003, **Jean Delvare** (now with SUSE) became the lead maintainer of lm-sensors. Jean is a well-known kernel developer (maintainer of Linux hardware monitoring *hwmon* and DMI subsystems) and was a primary upstream maintainer of lm-sensors from the 2000s through the 2010s. He is credited with significant portions of the code (e.g. `sensors-detect`) and took over project leadership. In the mid-2000s, **Mark M. Hoffman** also contributed and maintained parts of the project (he appears as a maintainer in 2005–2006).

- **2010s to present:** **Guenter Roeck** (a kernel developer involved in hwmon) became a key contributor and co-maintainer. Guenter was active in migrating the project infrastructure (e.g. commenting on moving the website in 2016) and is listed alongside Jean Delvare as a maintainer of Linux’s hardware monitoring subsystem. The official lm-sensors code was moved to GitHub under the **lm-sensors** organization. Jean Delvare continued to oversee the project (committing patches and doing releases) through at least 2019. For example, Jean is noted as one of the “main developers” of lm-sensors and was still authoring fixes in 2020. Guenter Roeck and other contributors (e.g. **Axel Thimm** handling infrastructure, and **Aurelien Jarno** from Debian) have also participated, but as of late 2010s Jean Delvare was the primary upstream maintainer.

---

### 📉 Current Project Activity Status

**Is lm-sensors actively maintained?** – The original lm-sensors repository on GitHub has seen **little activity since the last official release (3.6.0 on October 18, 2019)**. No new version releases have occurred after 3.6.0. Maintainer engagement on the GitHub project has been minimal in recent years. For instance, users have noted that issues have remained open for long periods with no response. In one case (Raspberry Pi sensors issue), a user commented that **“it’s now more than 2 years”** with no action from maintainers. Another community member bluntly stated in early 2024 that the upstream project is “pretty much dead since 3 years by now”, as there had been no commits or issue closures in that time frame.

Several Linux distributions and contributors have concluded that upstream development is stagnant. In mid-2024, Gentoo maintainers noted **“upstream lm-sensors has had no commits for three years”**. Likewise, Arch Linux maintainers observed that upstream hadn’t merged changes in a long time. While the GitHub repo still exists, the lack of recent maintainer activity indicates that the original project is in **maintenance mode or largely unmaintained** at this point.

> *(Note: The hardware monitoring drivers in the Linux kernel are actively maintained by kernel developers, but the lm-sensors user-space tools and library have not seen upstream updates in recent years.)*

---

### 📦 Distribution Packaging and Upstream Links

Major Linux distributions continue to ship lm-sensors, often with distro-specific patches, due to its importance for system monitoring. The package metadata in those distros sheds light on who is caring for lm-sensors:

#### Debian/Ubuntu:
- Maintainer: **Aurelien Jarno** (`aurel32@debian.org`)
- The Debian package was originally introduced by Stephen Crowley.
- Based on upstream `3.6.0` + Debian patches.
- Aurelien participates in upstream GitHub discussions and occasionally patches issues.

#### Fedora/RHEL:
- Maintainer: **Jaromir Capik** (Red Hat)
- Maintains *lm_sensors* in Fedora since at least 2012.
[... cut ...]
- The **original lm-sensors** project is **no longer actively maintained**.
- **Jean Delvare** and **Guenter Roeck** were the last primary maintainers.
- Linux distros continue shipping lm-sensors, patching issues locally.
- The **[SUSE fork by Michal Suchánek](https://github.com/hramrach/lm-sensors)** is currently the most active and maintained version.
- Distributions like **Gentoo** and **openSUSE** have already adopted it.
- No true replacement exists, but Prometheus/IPMI provide partial alternatives.

hramrach, 2025-06-27:
Active fork, eh. Perhaps most recently active fork.

I tried to modernize the library somewhat to make it easier to use with scripting languages and it's mostly done.

There is the problem it breaks API and I did no decide how to approach that breakage.
Also I got distracted by other things before finishing this.

Generally there is this problem the lm-sensors is two projects in one.

There is the tool and the library, and this is mostly fine.

The other part is the collection of presets for various boards that have sensors on them. This is very hard to maintain for many boards in one place. I have seen the maintainers of the original repository looking up sensor chip specifications and contesting preset submissions that are outside of the value range the chip can report. Is there anyone who will do this in the future?

For the platforms I am interested in this data should be stored in the device tree, and maintained by the people who are familiar with the hardware, in the device tree that describes their hardware. That keeps the data out of lm-sensors, and likely makes it feasible to maintain. For legacy platforms that don't provide this data in any intelligible form I do not see any good solution.

ilf, 2026-03-06:
The last commit was over five years ago: https://github.com/lm-sensors/lm-sensors/commits/master/

If the maintainer(s) want to abandon this project: Can they say so?

Does anyone else want to stand up?